gdsfactory/gdsfactory #656: get_netlist raises Found multiple connections

joamatab, 2022-09-04:
```
import gdsfactory as gf

if __name__ == "__main__":
    c = gf.components.switch_tree(bend_s=None)
    n = c.get_netlist_recursive()
```

get_netlist raises Found multiple connections

```
ValueError: Found multiple connections at (-5.5, 0.0):['e1', 'compass_1,e1', 'compass_2,e1', 'compass_3,e1']
```

@tvt173

tvt173, 2022-09-04:
it looks like your issue is coming from trying to netlist `via_stack_02e7e014`

looks like you have three instances of `compass` right on top of each other. all of them declare an `e1` port in the same location, as does the top level. i would challenge why you are constructing this component this way. i don't think the ports of `compass` should actually be `electrical`. i think only the top level component should have an electrical port, and the compass ports should simply be a `placement` type. and we should exclude `placement` ports from being netlisted. that's at least how i would manage this situation

[image]

joamatab, 2022-09-05:
I see, sending a PR that makes the compass default `port_type='placement'`

How about this case?
```
"""FIXME."""

import gdsfactory as gf

yaml = """
instances:
    mmi1:
      component: mmi1x2
      settings:
        width_mmi: 4.5
        length_mmi: 10
    mmi2:
      component: mmi1x2
      settings:
        width_mmi: 4.5
        length_mmi: 5

    straight:
        component: straight

placements:
    mmi2:
        x: 100
        mirror: True

    straight:
        x: 40
        y: 40

routes:
    route_name1:
        links:
            mmi1,o3: mmi2,o3
    route_name2:
        links:
            mmi1,o2: straight,o1
    route_name3:
        links:
            mmi2,o2: straight,o2

ports:
    o1: mmi2,o1
    o2: mmi2,o1
"""

if __name__ == "__main__":

    mzi = gf.read.from_yaml(yaml)
    mzi.show()
    n = mzi.get_netlist()
    # mzi.plot()

```

tvt173, 2022-09-05:
you are trying to promote the same lower-level port to two different ports on the top level. in your `port` definition, I think you should have `o1: mmi1,o1` and `o2: mmi2,o1` . then your example works

tvt173, 2022-09-05:
I would also put `"placement"` ports in the default `exclude_port_types` for `get_netlist()`. that will ensure that those ports do not attempt to get netlisted at all

PR #666 (2022-09-05): get_netlist ignores ports with port_type='placement'
get_netlist ignores ports with port_type='placement'
closes #656
closed 2022-09-05 by joamatab