gdsfactory/gdsfactory #659: Import error

thomasdorch, 2022-09-04:
I'm getting circular import errors when running layers.py as a script:
```
Traceback (most recent call last):
  File "/home/thomas/layout/gdsfactory/gdsfactory/layers.py", line 20, in <module>
    from phidl.device_layout import _CSS3_NAMES_TO_HEX
  File "/home/thomas/mambaforge/envs/gdsfactory/lib/python3.10/site-packages/phidl/__init__.py", line 1, in <module>
    from phidl.device_layout import (
  File "/home/thomas/mambaforge/envs/gdsfactory/lib/python3.10/site-packages/phidl/device_layout.py", line 41, in <module>
    from copy import deepcopy as _deepcopy
  File "/home/thomas/layout/gdsfactory/gdsfactory/copy.py", line 4, in <module>
    from phidl.device_layout import CellArray
ImportError: cannot import name 'CellArray' from partially initialized module 'phidl.device_layout' (most likely due to a circular import) (/home/thomas/mambaforge/envs/gdsfactory/lib/python3.10/site-packages/phidl/device_layout.py)
```
Is this something others are seeing as well?

tvt173, 2022-09-04:
can't say I have, but `CellArray` is a gdspy class anyways, right? shouldn't we import from gdspy directly @joamatab ?

thomasdorch, 2022-09-04:
@tvt173 It looks like `CellArray` is a gdspy class, but the `CellArray` here is the one from phidl, which is subclassed from the gdspy one and adds some additional methods.

joamatab, 2022-09-04:
great catch Thomas!

The issue is coming from `gdsfactory.copy` 

sending a PR for this

PR #660 (2022-09-04): move gdsfactory.copy to Component.copy
closes #659
PR #1261 (2023-02-09): fix settings
fixes #659 

thank you @gdspaul
closed 2022-09-04 by joamatab