gdsfactory/gdsfactory #678: Automatic generation of collateral for design tools.

joamatab, 2022-09-08:
It would be great to have automatic generation of collateral for design electronics tools.

- [ ] LEF
- [ ] DEF
- [ ] LIB 
- [ ] symbol view or dummy verilog

Digital tools need (LEF, DEF, dummy verilog)

Analog (symbol view and GDS (pins and output)

Ideas:

- open access export

tvt173, 2022-09-13:
i misspoke previously. oascript used to be open source. it now appears to require si2 membership to get access to the latest version.
https://si2.org/oascripting-solutions/

however, an old version (2014) can be found here, on gitlab
https://gitlab.com/Fatsie_EDA/oascript

joamatab, 2022-09-19:
this package also looks nice

https://github.com/cascode-labs/virtue

@curtisma

tvt173, 2022-09-22:
this tool is in rust, but it seems it can read/write lef. also a candidate for an alternate backend to gds
https://github.com/dan-fritchman/Layout21

joamatab, 2023-10-24:
we can do some of this with the vlsir plugin

https://gdsfactory.github.io/gplugins/notebooks/vlsir_netlist.html

closed 2023-10-24 by joamatab