gdsfactory/gdsfactory #681: Do we have a spice netlist generator in GDSfactory

alibillalhammoud, 2022-09-08:
Currently with OpenFASOC we are using GDS factory especially in PMU-gen. The issue we are having is that GDS factory does not have the capability to dump a netlist (instead it outputs a GDS). We need a pre-PEX netlist for LVS and running spice simulations for early on design exploration. OpenROAD has a “write cdl” function which we are using for this purpose in other generators such as the temp-sense-gen. We would like similar functionality in GDS factory.

joamatab, 2022-09-08:
Hi Ali,

all gdsfactory components have a  YAML "netlist" that you can get using `Component.get_netlist()`

as @danchitnis  pointed out, our YAML netlist is more of layout representation of the circuit, rather an a netlist

it would be great if you could add this feature to gdsfactory as a PR

msaligane, 2022-09-09:
@elon-J regarding the LVS issues in the PMU. Maybe you can take a look  at:

> all gdsfactory components have a YAML "netlist" that you can get using Component.get_netlist()

@joamatab when you say gdsfactory components, do you mean pcells?

joamatab, 2022-09-09:
Hi Mehdi,

I mean Component 
I am also assuming that the Pcell that returns a Component has been correctly constructed
see https://gdsfactory.github.io/gdsfactory/notebooks/11_get_netlist.html

the `get_netlist` is what we use for circuit simulations based on Sparameters
see https://gdsfactory.github.io/gdsfactory/notebooks/plugins/sax/sax.html

Something we need to do is a Pyspice version of the SAX notebook
where we extract connectivity of the cells, map each cell with a spice model, and then solve the circuit

@msaligane 
@mabrains
@mkghub

https://github.com/gdsfactory/skywater130/issues/16

joamatab, 2022-09-10:
Some ideas to make this happen:

- [ ] associate standard pdk cells to spice models (for example skywater pdk)
- [ ] write a spice netlist translator from YAML netlist to SPICE netlist `Component_get_netlist_spice()`

This would allow bulding spice models for PDK components and doing electronics photonics co-simulation

@tvt173 
@HelgeGehring 
@flaport 
@PetePupalaikis

danchitnis, 2022-09-10:
Currently waiting on few other projects on Skywater130 to become open source and converge, so hopefully we will have pcells soon. (Also processing raw measurements data to verify the pcell equations)

tvt173, 2022-09-13:
hi @joamatab , a few comments

1. I think we should allow the yaml netlist spec to represent a source, logical-netlist. I think the key modification needed here would be to add a `nets` property to the netlist, in which we list logical links, then make `placements` and `routes` optional, so that they can be defined later. You can imagine then, this can be done either manually or through automated algorithms, and combined with the "source netlist" file to instantiate a layout.
2. Implementing (1) will enable us to make an SDL-like flow, LVS, and pre-layout simulation, which I think would all be great to have. If you have SPICE output, then you could also run LVS or netlist-netlist comparisons on traditional tools with the source vs extracted netlists.
3. I agree that SPICE netlist output would be very nice to have. But how would most people want their SPICE netlists to look? Pure electrical? Pure optical? Both electrical and optical? I think it would nice to be able to configure which of these "views" you would like to get. You could probably use the options in the current `get_netlist()` function to ignore certain port types when netlisting, i.e. to get an electrical-only view of the circuit.

danchitnis, 2022-09-13:
SPICE netlist is only for electrical simulation. Optical simulation is based on FDTD, and I am not sure if there is a standardized netlist among the optical simulation software or if "netlist" makes sense for optical components.

joamatab, 2022-09-13:
an optical netlist represents connectivity of optical elements, and we use Sparameters of each component and a linear circuit solver to solve the overall circuit Sparameters

when you extract the netlist for SPICE electrical simulations, do we need to extract and include the resistance/inductance/capacitance that the wires add to the circuit?

simbilod, 2022-09-13:
> 3. I agree that SPICE netlist output would be very nice to have. But how would most people want their SPICE netlists to look? Pure electrical? Pure optical? Both electrical and optical? I think it [...]

Eventually an electrical+optical netlist would make sense to model electrically-reconfigurable optical components and detectors

danchitnis, 2022-09-13:
> an optical netlist represents connectivity of optical elements, and we use Sparameters of each component and a linear circuit solver to solve the overall circuit Sparameters

@joamatab is this only for electro-optical interfaces or pure optical elements like waveguides etc?

> when you extract the netlist for SPICE electrical simulations, do we need to extract and include the resistance/inductance/capacitance that the wires add to the circuit?

netlist is not aware of the physical properties of wires. They have to be added in as RLCK elements usually via post-layout extraction and back annotations. So it depends on extracting the netlist from the schematic or layout.

joamatab, 2022-10-22:
We could leverage https://github.com/Vlsir/Vlsir 
to support both spice output and rust backend and spice netlist

@growly
@proppy

joamatab, 2022-11-17:
Some ideas for extracting from layout:
- add a flag to `Component.get_netlist(with_spice_format=False)` where we return YAML format by default but we can return a SPICE netlist
- Optional: use VLSIR to create an intermediate format for SPICE (optional) that can be used for ngspice and xyce
- run spice simulation with pyspice

joamatab, 2022-12-09:
nice workshop
https://www.youtube.com/watch?v=pJ7S_usjEps

@Karlsefni2012 
@tim-edwards

proppy, 2022-12-29:
Looking at the netlist for sky130 examples here:
https://gdsfactory.github.io/skywater130/notebooks/intro.html

```
{'connections': {},
 'instances': {'sky130_fd_sc_hd__a2111o_1_1': {'component': 'sky130_fd_sc_hd__a2111o_1',
   'settings': {}},
  'sky130_fd_sc_hd__a311oi_4_1': {'component': 'sky130_fd_sc_hd__a311oi_4',
   'settings': {}},
  'straight_1': {'component': 'straight',
   'info': {'length': 9.9, 'width': 0.2},
   'settings': {'cross_section': {'layer': [68, 20],
     'width': 0.2,
     'offset': 0.0,
     'radius': 10.0,
     'width_wide': None,
     'auto_widen': False,
     'auto_widen_minimum_length': 200.0,
     'taper_length': 10.0,
     'bbox_layers': [],
     'bbox_offsets': [],
     'cladding_layers': ['DEVREC'],
     'cladding_offsets': [0.0],
     'sections': [],
     'port_names': ['o1', 'o2'],
     'port_types': ['optical', 'optical'],
     'min_length': 0.01,
     'start_straight_length': 0.01,
     'end_straight_length': 0.01,
     'snap_to_grid': None,
     'info': {},
     'name': None,
     'add_center_section': True},
    'length': 9.9}},
  'straight_2': {'component': 'straight',
   'info': {'length': 14.905, 'width': 0.2},
   'settings': {'cross_section': {'layer': [68, 20],
     'width': 0.2,
     'offset': 0.0,
     'radius': 10.0,
     'width_wide': None,
     'auto_widen': False,
     'auto_widen_minimum_length': 200.0,
     'taper_length': 10.0,
     'bbox_layers': [],
     'bbox_offsets': [],
     'cladding_layers': ['DEVREC'],
     'cladding_offsets': [0.0],
     'sections': [],
     'port_names': ['o1', 'o2'],
     'port_types': ['optical', 'optical'],
     'min_length': 0.01,
     'start_straight_length': 0.01,
     'end_straight_length': 0.01,
     'snap_to_grid': None,
     'info': {},
     'name': None,
     'add_center_section': True},
    'length': 14.905}},
  'wire_corner_1': {'component': 'wire_corner',
   'info': {'length': 0.2},
   'settings': {'cross_section': {'function': 'cross_section',
     'settings': {'add_pins': {'function': 'add_pins_siepic',
       'settings': {'pin_length': 0.002}},
      'add_bbox': {'function': 'add_bbox_siepic'},
      'cladding_layers': ['DEVREC'],
      'cladding_offsets': [0],
      'layer': [68, 20],
      'width': 0.2}}}}},
 'placements': {'sky130_fd_sc_hd__a2111o_1_1': {'x': 0.0,
   'y': 0.0,
   'rotation': 0,
   'mirror': 0},
  'sky130_fd_sc_hd__a311oi_4_1': {'x': 15.0,
   'y': 10.0,
   'rotation': 0,
   'mirror': 0},
  'straight_1': {'x': 0.23, 'y': 2.72, 'rotation': 90, 'mirror': 0},
  'straight_2': {'x': 0.33, 'y': 12.72, 'rotation': 0, 'mirror': 0},
  'wire_corner_1': {'x': 0.23, 'y': 12.72, 'rotation': 90, 'mirror': True}},
 'ports': {},
 'name': 'Unnamed_ebfccc23',
 'warnings': {'electrical': {'unconnected_ports': [{'ports': ['sky130_fd_sc_hd__a2111o_1_1,VGND',
      'sky130_fd_sc_hd__a2111o_1_1,VPWR',
      'sky130_fd_sc_hd__a311oi_4_1,VPWR',
      'sky130_fd_sc_hd__a311oi_4_1,VGND',
      'wire_corner_1,e1',
      'wire_corner_1,e2'],
     'values': [[0.23, 0.0],
      [0.23, 2.72],
      [15.235, 12.72],
      [15.235, 10.0],
      [0.23, 12.62],
      [0.33, 12.72]],
     'message': '6 unconnected electrical ports!'}]},
  'optical': {'unconnected_ports': [{'ports': ['straight_1,o1',
      'straight_1,o2',
      'straight_2,o1',
      'straight_2,o2'],
     'values': [[0.23, 2.72], [0.23, 12.62], [0.33, 12.72], [15.235, 12.72]],
     'message': '4 unconnected optical ports!'}]}}}
```
There doesn't seems to be any connectivity information in the `connections` json node atm, is that expected?

Do the ports need to be `connect()`'ed explicitly in addition of calling `c.add(route.references)`.

proppy, 2022-12-29:
trying magic `ext2spice` on the resulting gds:
```​
gds read test.gds
extract
ext2spice lvs
ext2spice cthresh 0
ext2spice
```
also trigger some interesting `Unknown layer/datatype errors:
```
Magic 8.3 revision 358 - Compiled on Fri Dec 23 23:19:13 UTC 2022.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
Input style sky130(): scaleFactor=2, multiplier=2
The following types are not handled by extraction and will be treated as non-electrical types:
    ubm 
Scaled tech values by 2 / 1 to match internal grid scaling
Loading sky130A Device Generator Menu ...
Using technology "sky130A", version 1.0.372-1-g8f01d8f
Warning: Calma reading is not undoable!  I hope that's OK.
Library written using GDS-II Release 6.0
Library name: library
Reading "Unnamed_ebfccc23".
Reading "wire_corner_aa3346ba".
Reading "straight_877558d1".
Reading "sky130_fd_sc_hd__a311oi_4".
Reading "straight_0c78da5a".
Reading "sky130_fd_sc_hd__a2111o_1".
Error while reading cell "straight_877558d1" (byte position 632): Unknown layer/datatype in boundary, layer=68 type=0
CIF file read warning: CIF style sky130(): units rescaled by factor of 5 / 1
Error while reading cell "straight_877558d1" (byte position 730): Unknown layer/datatype in path, layer=1 type=10
Error while reading cell "straight_0c78da5a" (byte position 16062): Unknown layer/datatype in boundary, layer=68 type=0
Error while reading cell "straight_0c78da5a" (byte position 16160): Unknown layer/datatype in path, layer=1 type=10
CIF file read warning: Input off lambda grid by 2/5; snapped to grid.
Please name the cell before extracting.
```

joamatab, 2022-12-29:
Hi proppy 

You need to connect at least two components using metal routes or connect components directly to each other

simbilod, 2022-12-29:
I have some work [on this branch](https://github.com/simbilod/gdsfactory/tree/spice) where I added an option to have multiple connections on a netlist, for instance like [the Spice notebook](https://github.com/simbilod/gdsfactory/blob/spice/docs/notebooks/plugins/spice/01_voltage_divider.ipynb) where I have a voltage divider. On main the connections must be 1-1.

In that branch I can also assign PySpice models to Components similarly to how we assign models to Component in SAX netlists. There is some more work needed in properly writing the netlist in PySpice format and defining a netlist with the right level of hierarchy.

RTimothyEdwards, 2022-12-29:
@proppy :  `68:0` is a production mask layer number;  it's not supposed to be used for design.  For design shapes, you should be using the drawing purpose `68:20`.  Also:  layer `1:10` is not anything defined by SkyWater in any process in any file that I can find.

proppy, 2022-12-30:
> You need to connect at least two components using metal routes or connect components directly to each other

Looking at the example at https://gdsfactory.github.io/skywater130/notebooks/intro.html I was under the impression that the two component were already routed together thru the call to:
```
route = gf.routing.get_route_electrical(g1.ports['VPWR'], g2.ports['VPWR'], cross_section=st.xs_metal1)
c.add(route.references)
```

Does that mean that one need to also call, to get fully connected components?
```
g1.connect("VPWR", g2.ports['VPWR'])
```

proppy, 2022-12-30:
> for instance like [the Spice notebook](https://github.com/simbilod/gdsfactory/blob/spice/docs/notebooks/plugins/spice/01_voltage_divider.ipynb) where I have a voltage divider.

Thanks for sharing, that's awesome! It would be good to try to submit together a small design generated thru this flow to https://efabless.com/open_shuttle_program/.

> In that branch I can also assign PySpice models to Components similarly to how we assign models to Component in SAX netlists

Would this association ultimatly ends up in the sky130 gdsfactory components at https://github.com/gdsfactory/skywater130 ?

proppy, 2022-12-30:
> @proppy : 68:0 is a production mask layer number; it's not supposed to be used for design. For design shapes, you should be using the drawing purpose 68:20

Thanks for pointing that out, sounds like something that should be fixed in https://github.com/gdsfactory/skywater130/blob/f537582f5fe4b6bf96ed6fad4d40ee3ee418740f/sky130/tech.py#L9-L10, filed https://github.com/gdsfactory/skywater130/issues/51

>  Also: layer 1:10 is not anything defined by SkyWater in any process in any file that I can find.

Seems like it comes from
https://github.com/gdsfactory/skywater130/blob/f537582f5fe4b6bf96ed6fad4d40ee3ee418740f/sky130/layers.py#L437-L438, @joamatab those were added in https://github.com/gdsfactory/skywater130/commit/13f2d92032d8ba87247b10dd8d7b188801c3e3b2 do you have more context about this change?

joamatab, 2022-12-30:
Hi Proppy,

yes, layer `68/0` and `1/10` were layers that we used for using an external netlist extractor from another open source PDK

https://github.com/gdsfactory/gdsfactory/pull/1057
removes them from gdsfactory strip

and 
https://github.com/gdsfactory/skywater130/pull/52

removes them from skywater routing layers

closed 2023-09-05 by joamatab