gdsfactory/gdsfactory #685: improve A* router

joamatab, 2022-09-09:
```
import gdsfactory as gf

c = gf.Component("get_route_astar")
w = gf.components.straight()
left = c << w
right = c << w
right.move((100, 80))

obstacle = gf.components.rectangle(size=(100, 10))
obstacle1 = c << obstacle
obstacle2 = c << obstacle
obstacle1.ymin = 40
obstacle2.xmin = 25

port1 = left.ports["o2"]
port2 = right.ports["o2"]

routes = gf.routing.get_route_astar(
    component=c,
    port1=port1,
    port2=port2,
)
c.add(routes.references)
c.show()
```

[image]

```
import gdsfactory as gf

cross_section = "metal1"

c = gf.Component("get_route_astar")
w = gf.components.straight(cross_section=cross_section)

left = c << w
right = c << w
right.move((100, 80))

obstacle = gf.components.rectangle(size=(100, 10), layer="M1")
obstacle1 = c << obstacle
obstacle2 = c << obstacle
obstacle1.ymin = 40
obstacle2.xmin = 25

port1 = left.ports["e2"]
port2 = right.ports["e2"]

routes = gf.routing.get_route_astar(
    component=c, port1=port1, port2=port2, cross_section="metal1", resolution=5
)
c.add(routes.references)
c.show()
```

```
"""FIXME.

We need to add an option to avoid specific layers
"""

if __name__ == "__main__":
    import gdsfactory as gf

    c = gf.Component()

    rect1 = c << gf.components.rectangle()
    rect2 = c << gf.components.rectangle()
    rect3 = c << gf.components.rectangle((2, 2), layer=(2, 0))
    rect2.move(destination=(8, 4))
    rect3.move(destination=(5.5, 1.5))

    port1 = gf.Port(
        "o1", 0, rect1.center + (0, 3), cross_section=gf.get_cross_section("strip")
    )
    port2 = port1.copy("o2")
    port2.orientation = 180
    port2.center = rect2.center + (0, -3)
    c.add_ports([port1, port2])
    route = gf.routing.get_route_astar(c, port1, port2, radius=0.5, width=0.5)
    c.add(route.references)

    c.show(show_ports=True)

```

@SkandanC

SkandanC, 2022-09-09:
Hello @joamatab,

Thanks for letting me know. I will start working on this right away.

For the first example layout, I made a few minor adjustments, and changing the `resolution` parameter to 10 (a higher resolution minimizes turns/the pixelate-y effect), and changing the radius to 5 gives me this. Admittedly, it is not ideal, but I feel it is better than before. The main issue is that the A* algorithm does not optimize for minimal turns. Modifying it to do this would fix the issues in the first two examples. Could you explain what you mean by "avoid specific layers" in the third example?

[image]

joamatab, 2022-09-09:
nice, this looks much better, looking forward to your PR

it would be nice to add an optional `List[LayerSpec]` of layers that it needs to avoid, for example, 

metal1 only needs to avoid shapes in  metal1 , and does not need to avoid `waveguide` layer

i would use `c.get_polygons(layers)` so we only compute the bbox for the important layers that we need to avoid

SkandanC, 2022-09-09:
Okay, I will look into that.

One way to minimize number of turns I think is to compute 4 or 5 shortest paths, sort them by number of turns and pick the one with the least number of turns. I will see if there are other options.

SkandanC, 2022-09-09:
I added an `avoid_layers` option and `distance` option, and there is a significant improvement.

[image]

Doing 
```
cross_section = gf.get_cross_section("metal1", width=3, radius=1)

routes = get_route_astar(component=c, port1=port1, port2=port2, avoid_layers=[cross_section.layer], cross_section=cross_section, resolution=5, distance=5.5)
``` 
for the second example gives this:
[image]

There are still a large number of turns, I will work on improving that.

joamatab, 2022-09-09:
nice :tada: 

looks great Skandan

SkandanC, 2022-09-10:
I added a cost addition to the heuristic function in the algorithm. Essentially, everytime it detects a turn, it is penalized for it, and so it tries to avoid as many turns as possible. Here are the updated results:

[image]
[image]

closed 2022-09-10 by joamatab