gdsfactory/gdsfactory #692: astar router improvements

joamatab, 2022-09-10:
How can we fix the Astar router for the code below?

```
import gdsfactory as gf

c = gf.Component("get_route_astar")
w = gf.components.straight()
left = c << w
right = c << w
right.move((100, 80))

obstacle = gf.components.rectangle(size=(100, 10))
obstacle1 = c << obstacle
obstacle2 = c << obstacle
obstacle1.ymin = 40
obstacle2.xmin = 25

port1 = left.ports["o2"]
port2 = right.ports["o2"]

routes = gf.routing.get_route_astar(
    component=c,
    port1=port1,
    port2=port2,
)
c.add(routes.references)
c.show()
```

@SkandanC

SkandanC, 2022-09-10:
Doing 
```
routes = gf.routing.get_route_astar(
    component=c,
    port1=port1,
    port2=port2,
    resolution=5,
    distance=5.5,
    radius=5,
)
```
gives me this result in ~0.14 secs.

[image]

The resolution decides how many "leaps/hops" the algorithm has to do. For a layout like this, where the default resolution (1 micron) is much larger than the distance between the obstacles (~15+ microns), it has to step through too many points and that takes a long time. Increasing the resolution to about 5 microns fixes it (for this layout).

joamatab, 2022-09-10:
perfect, thank you Skandan,

i added your comments to the notebook for documentation

im also having issues with separation

```
import gdsfactory as gf

c = gf.Component("get_route_astar")
cross_section = "metal1"
w = gf.components.straight(cross_section=cross_section)

left = c << w
right = c << w
right.move((100, 80))

obstacle = gf.components.rectangle(size=(100, 10), layer="M2")
obstacle1 = c << obstacle
obstacle2 = c << obstacle
obstacle1.ymin = 40
obstacle2.xmin = 25

port1 = left.ports["e2"]
port2 = right.ports["e2"]

routes = gf.routing.get_route_astar(
    component=c,
    port1=port1,
    port2=port2,
    cross_section="metal1",
    resolution=5,
    distance=5,
)
c.add(routes.references)
c.show()
```

[image]

SkandanC, 2022-09-10:
```
routes = gf.routing.get_route_astar(
    component=c,
    port1=port1,
    port2=port2,
    cross_section="metal1",
    resolution=10,
    distance=5
)
```

[image]
 I increased the resolution to make the algorithm not take hops too close to the obstacles. I will see what I can do about the part near the output port.

SkandanC, 2022-09-10:
```
routes = gf.routing.get_route_astar(
    component=c,
    port1=port1,
    port2=port2,
    cross_section="metal1",
    resolution=10,
    distance=7,
)
```
Increasing the distanc to 7 fixed it.

[image]

joamatab, 2022-09-10:
I see, it's still touching on one side :)

[image]

closed 2022-09-10 by joamatab

SkandanC, 2022-09-10:
That is true. But, there isn't enough space there between the port and the obstacle to resolve the `distance` parameter there.

joamatab, 2022-09-10:
Yes, you are right, it looks great 

A more useful router would be a global router, I added some links to the issue #57