gdsfactory/gdsfactory #713: TCAD improvements

simbilod, 2022-09-15:
This issue tracks current limitations to TCAD simulations and possible improvements.

The PIN waveguide solver is a good start to combining gdsfactory with semiconductor solvers, and is OK to get order-of-magnitude estimates in a reverse-bias regime where there is little current flowing. However more needs to be done to increase usability:

1) Improved physics

Currently DEVSIM is used by loading the "simple_physics" from the base package. This only considers constant mobility, and simple SRH recombination. We can make our own modeling:

- [ ] Adapt https://github.com/devsim/devsim_bjt_example/blob/main/simdir/physics/new_physics.py, which adds
 * Quasi Fermi levels with Boltzmann statistics
 * Arora mobility (charged impurity scattering)
 * Caughey Thomas mobility (velocity saturation)
- [ ] Add avalanche breakdown
- [ ] Add hydrodynamic models
- [ ] Add local temperature effects
- [ ] Add surface mobility models

2) Generate simulation from components

- [ ] Generate a basic 2D FEM mesh from a gdsfactory cross_section
- [ ] Implement adaptive mesh refinements
- [ ] Generate a 3D FEM mesh from a gdsfactory components

3) Improved doping profiles

Until we have a process simulator that can model implantation and annealing, we could get slightly better doping models by assuming distributions (Gaussians, Pearsons) instead of sharp interfaces.

- [ ] Add analytical expressions for implantation profiles as a function of depth in terms of implant energy, using tables for depth and straggle

simbilod, 2022-09-20:
2a addressed in #729

smartalecH, 2022-09-30:
I'm interested in helping with this. I'll reach out once I get more bandwidth.

simbilod, 2022-09-30:
2c begun addressing in #745

tcaduser, 2022-10-09:
There are some @devsim implementations for the TCAD models listed.  However, it would require additional work to get them packaged together.

> This issue tracks current limitations to TCAD simulations and possible improvements.
>
> The PIN waveguide solver is a good start to combining gdsfactory with semiconductor solvers, and is OK to get [...]

For low current situations, extended precision 128-bit precision is available in devsim.  Mesh refinement may be able to help this as well.

> 
> Currently DEVSIM is used by loading the "simple_physics" from the base package. This only considers constant mobility, and simple SRH recombination. We can make our own modeling:
> 
> * [ ]  Add oxide regions w/ surface recombination
> * [ ]  Adapt https://github.com/devsim/devsim_bjt_example/blob/main/simdir/physics/new_physics.py, which adds
> 
> * Quasi Fermi levels with Boltzmann statistics
> * Arora mobility (charged impurity scattering)
> * Caughey Thomas mobility (velocity saturation)
> 
> * [ ]  Add avalanche breakdown

A @devsim user has contributed a sample implementation here:
https://forum.devsim.org/t/avalanche-breakdown/26

> * [ ]  Add hydrodynamic models
> * [ ]  Add local temperature effects
> * [ ]  Add surface mobility models

A MOSFET surface mobility model implementation is packaged here:
https://github.com/devsim/devsim_3dmos

> 
> 2. Generate simulation from components
> 
> * [x]  Generate a basic 2D FEM mesh from a gdsfactory cross_section
> * [ ]  Implement adaptive mesh refinements

A mesh refinement approach with respect to 2d simulation and Gmsh is packaged here:
https://github.com/devsim/devsim_misc/tree/main/refinement

Using a commercial mesher (commercial version of Cubit), a refinement strategy was implemented for the devsim_3dmos project above.

> * [x]  Generate a 3D FEM mesh from a gdsfactory components
> 
> 3. Improved doping profiles
> 
> Until we have a process simulator that can model implantation and annealing, we could get slightly better doping models by assuming distributions (Gaussians, Pearsons) instead of sharp interfaces.

1D numerical profiles where used for doping in the devsim_3dmos example above.

For analytical profiles, the `erf` and `erfc` functions are available.  In addition, arbitrary functions can be added using this command.

https://devsim.net/CommandReference.html#devsim.register_function

> * [ ]  Add analytical expressions for implantation profiles as a function of depth in terms of implant energy, using tables for depth and straggle

simbilod, 2022-10-10:
Thank you @tcaduser , this is very helpful! I'm currently reworking the meshing because I was not happy with my initial implementation. 

The goal is to programmatically be able to generate GMSH 3D volumetric meshes (or 2D cross-section) from gdsfactory components (so a GDS) and a gdsfactory layerstack. Then I want it to be easy to pass the various volumes and interfaces to a FEM solver like DEVSIM. Maybe it's something your users over at the DEVSIM forums might find useful when I get it working.

simbilod, 2022-11-08:
I am now satisfied with automated 2D meshing, see #826 

I have started porting the technique to 3D, and integrating DEVSIM more seamlessly

simbilod, 2023-01-04:
Meshing has seen a lot of progress, see https://github.com/gdsfactory/gdsfactory/pull/1065 and https://github.com/gdsfactory/gdsfactory/pull/1074

joamatab, 2023-09-13:
closing this 

feel free to reopen it in https://github.com/gdsfactory/gplugins

closed 2023-09-13 by joamatab