gdsfactory/gdsfactory #719: Allow for different mesh size in x and y in tidy3d

mdecea, 2022-09-15:
Right now, it seems like the tidy3d simulations in `modes.py` only allow for uniform grid with the same number of cells in x and y., which is pretty inflexible.

Important note: although in the argument description in the docstring it says that resolution is the number of pixels/um, that's not true! Resolution is indeed the total number of cells in the x and y directions. 

I want to request for at least the possibility of allowing different number of cells in the x and y directions. I have it implemented in my local copy of gdsfactory, so if you agree I can just create a PR. Just want to ask first because it is likely that some code will have to change as I have it implemented taking as parameters `resolution_x` and `resolution_y` instead of `resolution` (for example `get_simulation_xsection.py` will for sure have to change.

simbilod, 2022-09-15:
I'm the only one using `get_simulation_xsection.py`, so it's not a problem to change it for me

simbilod, 2022-09-15:
You could also make `resolution_x` and `resolution_y` new `Optional` parameters that override `resolution` if set

closed 2022-09-16 by joamatab