gdsfactory/gdsfactory #723: Simpler install

joamatab, 2022-09-18:
How can we simplify the installation process?

Options

- executable window anaconda on packager. Ship anaconda environment, use constructor to create a conda installer, or mambstructor for mamba 

@tvt173 
@HelgeGehring 
@simbilod 
@flaport 
@thomasdorch

joamatab, 2022-09-19:
We have now releases available with installers for windows, mac and linux
https://github.com/gdsfactory/gdsfactory/releases

thanks to constructor
https://github.com/conda/constructor/blob/main/CONSTRUCT.md

we need to add some sort of menu, launcher, shortcuts

one idea is using anaconda-navigator
https://docs.continuum.io/anaconda/navigator/

simbilod, 2022-09-20:
On Windows:

After downloading, the program shows up in Start menu:
[image]

The provided Spyder has the package:
[image]

There is also an Anaconda Navigator (albeit not under gdsfactory folder) that crashed for me on launch:
[image]

simbilod, 2022-09-20:
It would also be nice if there was a quick way to load the docs after a fresh download

simbilod, 2022-09-20:
Navigator works now, but it does not show up under the "gdsfactory" program (it is under anaconda)

joamatab, 2022-09-21:
is it worth having it as a menu? 

can you launch jupyterlab and spyder from it?

or shall we remove it and upgrade to python3.10?

simbilod, 2022-09-21:
Yes jupyterlabs and spyder can both be launched with the right environments from there, which is nice. If you are not a regular Python user and want to use gdsfactory this is a nice option

simbilod, 2022-09-21:
[image]

joamatab, 2022-09-22:
This looks great, maybe as part of the post-install script we can also git clone the gdsfactory repo into your desktop or home folder, so you can get started with the tutorial,

what do you think?

joamatab, 2022-09-26:
Hi,

does any of the users on windows see the gdsfactory on their desktop after running the installer?
It would also be great to create desktop shortcuts on windows

What do you think of using anaconda navigator to update gdsfactory?

Another option is to create a new launcher, GUI-like experience,
This would also allow us to do schematic driven flow
https://github.com/qiskit/qiskit-metal

@SkandanC 
@tvt173 
@simbilod

SkandanC, 2022-09-26:
@joamatab I do not see gdsfactory on my desktop after installation. Also, I feel like the installation time (at least on windows) is a little too long, perhaps it is my old laptop that's the issue, unless others have experienced this as well.

joamatab, 2022-09-26:
I have heard it's slow on windows, Mac and Linux is super fast compared to windows

tcaduser, 2022-09-26:
You may have a faster windows install speed by temporarily disabling the antivirus.

joamatab, 2022-09-26:
We are using conda constructor for building the installer
https://github.com/conda/constructor

For windows we use [install.bat](https://github.com/gdsfactory/gdsfactory/blob/master/install.bat) post install script
what do you get when you run this script by itself?

Did jax and jaxlib got installed too?

SkandanC, 2022-09-26:
@joamatab 
[image]

joamatab, 2022-09-28:
I see, If someone with a windows machine can fix the install.bat script it would be great

joamatab, 2022-10-02:
I found a VM for windows to debug the installer

https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/

SkandanC, 2022-10-07:
Turns out the error is because there's a version mismatch problem. The latest available version of jaxlib for Windows is 0.3.7, whereas the latest version of JAX is 0.3.21. So we can only have JAX 0.3.7 (released April 2022) for Windows users as of now 

@joamatab

SkandanC, 2022-10-07:
Actually, it turns out the latest version is available, the install command just needs to be modified.

joamatab, 2022-10-10:
Hi Skandan,

can you try again to see if it works?

it would be also great to add shorcuts on the desktop as part of the installer
https://github.com/conda/constructor/issues/573

joamatab, 2022-10-12:
We are not adding shortcuts yet

Did it installed the klayout macros?

joamatab, 2022-10-12:
If you uninstall the gdsfactory klayout integration install.bat should install it for you

The integration gives you the generic layer map and Component.show()

SkandanC, 2022-10-12:
Yep, it works, i'm able to do Component.show() after running the installer.

joamatab, 2022-10-12:
Awesome, thank You Skandan!

closed 2022-10-12 by joamatab