gdsfactory/gdsfactory #735: expand plugins

joamatab, 2022-09-26:
How could we expand gdsfactory to other use cases?

- RF
- analog design 

ideas: 

- pyspice plugin
- plugin for HFSS [see for example](https://github.com/Qiskit/qiskit-metal/tree/main/qiskit_metal/renderers/renderer_ansys)
- plugin for qiskit metal

@tula3and

simbilod, 2022-09-27:
@joamatab for the Spice plugin, we will need to rethink `get_netlist` because it currently restricts nodes to 2 connections only (which is too limiting for electrical netlists)

For instance, here I am trying to define a simple voltage divider:

[image]
[image]

Once we can label individual nodes, it should be easy to convert to a spice netlist, assigning Spice models between nodes conditioned on the component there. Then we can just run regular spice analyses.

tvt173, 2022-09-27:
hi @simbilod , you appear to be looking at an old version of gdsfactory. please pull the latest :)

that won't fully solve your issue, but it will take you a step closer! I submitted a PR a few weeks ago to rewrite `get_netlist()`, mainly for other reasons, but also with the issue you bring up in mind... `get_netlist()` now looks more like this:
[image]

the critical change i want to highlight, is that for each port type, the connections are now potentially parsed differently. so you can refactor to parse electrical connections differently than optical ones (which can be left as-is).

i would suggest refactoring to parse the electrical connections in a way that's more forgiving for electrical connections. it would be nice to also handle connections through vias.

also, `connections` in the netlist dictionary expects a 1:1 mapping of `inst1,port1: inst2,port2` which is a bit awkward, but i don't think _necessarily_ needs to be changed immediately... for example, if `inst3,port3` were also connected at that point, you could just add another entry `inst3,port3: inst2,port2` and that would convey the appropriate information (even if not concisely/clearly).

joamatab, 2023-09-05:
fixed it here

https://github.com/gdsfactory/gplugins/pull/80

closed 2023-09-05 by joamatab