gdsfactory/gdsfactory #747: EMEpy integration

smartalecH, 2022-09-30:
It would be good to add a plugin for [EMEpy](https://github.com/BYUCamachoLab/emepy) for streamlined EME calculations.

We have plans for GPU offloading (once we find a cuda compatible targeted eigensolver that works well with python) and adjoint optimization (using a clever mix of diff-then-disc and disc-then-diff).

(cc @hammy4815)

smartalecH, 2022-09-30:
Adding a plugin like this would actually be mutually beneficial:
* gdsfactory gets another solver
* EMEpy gets better corner case testing and validation

joamatab, 2022-09-30:
great idea,

having eme on gdsfactory would be great for modeling tapers, bends, adiabatic couplers ...

@tvt173 
@flaport

joamatab, 2022-11-08:
Another open sourced another EME plugin that would be great to integrate with gdsfactory
https://github.com/flaport/meow

thank you Floris!

@SkandanC 
@HelgeGehring 
@simbilod 
@thomasdorch

simbilod, 2022-12-19:
see #1015

joamatab, 2023-02-16:
I'll close this for now, PRs for simulation interfaces are always welcome

closed 2023-02-16 by joamatab