gdsfactory/gdsfactory #749: Meep adjoint optimization plugin / gmeep extension

SkandanC, 2022-09-30:
It would be nice to extend gmeep's capabilities to include Meep's adjoint optimization features. This way, we can optimize some of gdsfactory's primitive components such as mmi1x2, generate a gdsfactory component from them, and eventually gds files.

Things to do:
- [x] Figure out how to setup a Meep adjoint optimization simulation from gdsfactory components
- [ ] Figure out how to extract a gdsfactory Component object, and eventually a gds file, from the PNG files generated by Meep

For the latter, we can convert the PNG's to a numpy array, and then use gf.read.from_np to generate a gdsfactory component, as per @joamatab's suggestion. Or, if there's a way to create gds files straight from Meep, we can just turn those into a gdsfactory component.

thomasdorch, 2022-09-30:
I actually wrote a function to get gdsfactory components from meep sims for doing exactly this, accessing the epsilon grid from the simulation object:

```
def get_component_from_sim(
        sim: mp.Simulation,
        upscale_factor: int = 2,
        threshold_offset_from_max: float = 0.01,
        layer: gf.types.Layer = (1, 0)
):
    grid_resolution = upscale_factor * sim.resolution

    sim_center, sim_size = mp.visualization.get_2D_dimensions(sim, output_plane=None)
    xmin = sim_center.x - sim_size.x / 2
    xmax = sim_center.x + sim_size.x / 2
    ymin = sim_center.y - sim_size.y / 2
    ymax = sim_center.y + sim_size.y / 2

    Nx = int((xmax - xmin) * grid_resolution + 1)
    Ny = int((ymax - ymin) * grid_resolution + 1)

    xtics = np.linspace(xmin, xmax, Nx)
    ytics = np.linspace(ymin, ymax, Ny)
    ztics = np.array([sim_center.z])

    eps_data = np.real(sim.get_epsilon_grid(xtics, ytics, ztics, frequency=fcen))

    return gf.read.from_np(
        eps_data,
        nm_per_pixel=1e3 / grid_resolution,
        layer=layer,
        threshold=np.max(eps_data) - threshold_offset_from_max,
    )
```

Never got around to adding it to gdsfactory but feel free to use it as a starting point :)

SkandanC, 2022-10-02:
[image]
Managed to do it. Ty @thomasdorch for your help! This is after 72 iterations of the optimization, which probably is a little low, and so the structure may not be the best. But, it serves the purpose of demonstration.

Regular mmi1x2 for comparison:
[image]

joamatab, 2022-10-02:
Hi Skandan, 

this looks great,

can you send a link to the GDS?
There seems to be some self intersecting polygons that foundries won't accept?

have you made sure it passes some min width and min gap DRC rules using klayout? 
You can use 100nm for example both for min width and min gap

@smartalech
@flaport

smartalecH, 2022-10-02:
How are you specifying the design region, FOM, optimization algorithm, beta scheduling, constraints, etc from within gdsfactory? 

Or is the optimization done externally, and the geometry simply imported after the fact?

We have some contour extraction routines that have yielded numerous fab-friendly results. (cc @joelslaby).

SkandanC, 2022-10-02:
> Hi Skandan,
>
> this looks great,
>
> can you send a link to the GDS? There seems to be some self intersecting polygons that foundries won't accept?
>
> have you made sure it passes some min width and min [...]

@joamatab, I stopped the optimization before it could be done, because I didn't think my old laptop can handle more 😅. That explains why it has self - intersecting polygons and some features smaller than what foundries may accept. I will try to let it run for longer, and then send a link for the updated GDS file.

@smartalecH, as of now, the user defines the objective functions, FOM, optimization algorithm, etc. GDSFactory simply takes in the component, creates a Meep Simulation object, and then runs the optimization. Then the component can be generated from the Simulation object that is returned. I am sure there's a better way to do it, but I will need a couple of days to figure it out.

SkandanC, 2022-10-02:
[image]
I adjusted the threshold a bit and it looks better. 

[Here's a link to the GDS file.](https://drive.google.com/file/d/1q88A5JMH_0FPZpwBy51AB3gToWrY8xJ-/view?usp=sharing)

joamatab, 2022-10-03:
Looks great,

it has some DRC errors
[image]

and some strange things on the edges
[image]

maybe you need to simplify the contours

SkandanC, 2022-10-03:
Here is an updated pic after the optimization was complete (I ran it for 97 iterations).
[image]

[Here is the updated GDS file link](https://drive.google.com/file/d/1q88A5JMH_0FPZpwBy51AB3gToWrY8xJ-/view?usp=sharing)

I think if I play around with the optimization hyperparameters I can eventually get a geometry with no DRC errors.

smartalecH, 2022-10-03:
Note that the filters alone don't enforce DRC. You also need to add the corresponding optimization constraints that work with the filters. 

We don't have an end to end tutorial yet, but are working on it.

SkandanC, 2022-10-05:
I am almost done with this, and am nearly ready to open a PR.

However, when I try to run a test optimization run, I get this error:
[image]

This is my optimization object for reference:
[image]

Does anyone have any clues as to why this could be happening?

smartalecH, 2022-10-05:
What version of meep are you using?

Essentially the simulation hasn't been initialized right. There are some checks in the step routine that ensures everything is as expected.

smartalecH, 2022-10-05:
(Also you want more space between your PML and device)

smartalecH, 2022-10-05:
(Also your monitors and source lines are too narrow. They won't calculate the proper mode)

SkandanC, 2022-10-05:
Thanks Alec! I did some more playing around and found out that generating the `Simulation` object for the optimization directly from Meep, rather than from gdsfactory, seems to fix the problem. Perhaps I am using the `get_simulation` method from gmeep incorrectly.

I also adjusted the port margins and the cell size.

SkandanC, 2022-10-05:
Also, I am using Meep version 1.24.0

SkandanC, 2022-10-06:
Figured out what the issue was, the source generated for the Simulation object by gdsfactory couldn't be used in the optimization for some reason. I added a bit of code to generate a source that will, and it runs fine.

I am running this simulation one last time, and then I will add this to the gmeep docs, and open a PR @joamatab

SkandanC, 2022-10-07:
[image]

Final optimized structure. 

[image]

Still some areas are way too small, but playing around with the filter and constraint settings should fix it I think.

The FOM can be computed in the cost function:
[image]

The cost function, and objective arguments and functions are all defined by the user and passed in as arguments.

closed 2022-10-07 by SkandanC

smartalecH, 2022-10-07:
Very cool! Nice job, @SkandanC!