gdsfactory/gdsfactory #758: Large grids take a really long time to generate

mdecea, 2022-10-06:
Not totally sure if it's really a bug, but I'm trying to generate a (pretty large) grid of simple elements (eg, a grid of 1,000x1,000 squares to generate a lattice). 

It takes gdsfactory a really large amount of time to draw such a grid - for a 100x100 elements it takes about 2 minutes, and a 1000x1000 grid takes about 10 minutes.

Is there a way this can be faster?

mdecea, 2022-10-06:
Tried it with `add_array` and it was much faster.

The way I was doing it was the following:

```
base = gf.components.rectangle(size=(1.0, 1.0), layer='WG')
ar = gf.components.array(component=base, spacing=(2.0, 2.0), columns=1000, rows=1000)
c << ar
```

This took about 10 minutes in my PC.

tvt173, 2022-10-07:
interesting... since `gf.components.array()` simply creates a component, adds the array, then adds the ports, it must be the way adding the ports is done that is inefficient (when theres N * 1 million of them!). maybe we can profile and clean that code up a bit

closed 2022-10-09 by joamatab