gdsfactory/gdsfactory #774: dbr_tapered bug

SkandanC, 2022-10-14:
**Describe the bug**
dbr_tapered periodic structures are offset from the correct position if the duty cycle is not 0.5

**To Reproduce**
When duty cycle is 0.5:
[image]
When duty cycle is not 0.5:
```
dbr_tapered(length=5, period=0.85, dc=0.8, w1=0.4, w2=1, taper_length=5, fins=False, fin_size=(0.2, 0.05)).show()
```
[image]

**Expected behavior**
The periodic structures must be positioned in the "centre" of the taper + straight section

**Suggested fix**
A simple fix would be to rewrite the component to be similar to dbr, wherein the user passes in the number and widths of the periodic structures, and it creates an array of the periodic structures cells. 

I can open a PR with this fix if this sounds good @joamatab

SkandanC, 2022-10-14:
I modified by making it use `array` instead of manually adding the periodic structures. I will open a PR soon.

closed 2022-10-15 by SkandanC

joamatab, 2022-10-15:
Thank you Skandan