gdsfactory/gdsfactory #787: Meep improvements

joamatab, 2022-10-26:
- how can we plot refractive index colorbar on meep?
- how can we use plot2D when the simulation is 3d?

```
import gdsfactory.simulation.gmeep as gm
import gdsfactory.simulation as sim
import meep as mp
import gdsfactory as gf

nm = 1e-3

c = gf.components.mmi1x2(cross_section=gf.cross_section.nitride)
fig = c.plot()

# How can we plot the epsilon or refractive index?
sim = gm.get_simulation(c, is_3d=False)['sim']
sim.plot2D(output_plane=mp.Volume(size=sim.cell_size, center=mp.Vector3(0, 0, 0*nm)))

# This does not work
sim = gm.get_simulation(c, is_3d=True)['sim']
sim.plot2D()
```

@SkandanC 
@smartalech
@HelgeGehring

HelgeGehring, 2022-10-26:
For plotting a 2D slice of a 3D simulation one dimension of size needs to be zero, like
```
sim.plot2D(output_plane=mp.Volume(size=mp.Vector3(sim.cell_size.x, sim.cell_size.y, 0.), center=mp.Vector3(0, 0, 0 * nm)))
```

SkandanC, 2022-10-26:
[image]
This is how it says to do in the Meep documentation. Perhaos could be a starting poitn for us.

smartalecH, 2022-10-26:
> how can we use plot2D when the simulation is 3d?

@HelgeGehring is right. The idea is you need to pass a _plane_ to visualize (specified by a `Volume` object).

> how can we use plot2D when the simulation is 3d?

The output of `plot2D()` [returns the axis handle ](https://github.com/NanoComp/meep/blob/5fa12b499d4934ea8b352e3e8033762b34f6eb0c/python/visualization.py#L554)of the plot. You should be able to loop through this handle until you find the relevant `imshow` data structure, which you can then use to create a colorbar.

I don't have any examples offhand... the last time I actually did this was when I wrote the function 3+ years ago...

smartalecH, 2022-10-27:
> Would it be a good idea to do have something similar to gf.simulation.add_simulation_markers() but with a transparent box added which represents the PML? I feel like this could be useful in creating 3D scenes of Meep sim objects.

Sure that'd be great. Meep PRs are always welcome. 3D rendering with python has come a long way since we first added visualization support to meep.

joamatab, 2022-10-27:
Thank you Skandan, Alec and Helge

The colorbar seems now more of a meep issue than gdsfactory issue

https://github.com/NanoComp/meep/issues/2286

closed 2022-10-27 by joamatab