gdsfactory/gdsfactory #800: flayout improvements

joamatab, 2022-10-31:
Thanks to flayout and Thomas' PR we can now install gdsfactory inside klayout and instantiate all the generic cells.
This allows to convert any code driven gdsfactory pdk into a klayout PDK
https://github.com/gdsfactory/gdsfactory/pull/797

[ ] allow to modify Pcell values
[image]

[ ] convert gdspy paths into 
[image]

[ ] remember gdsfactory packages path. So it does not ask you for their location everytime

[x] show in menu
 - [ ] group all gdsfactory macro menus inside a `gdsfactory` menu

joamatab, 2022-11-01:
Can someone test it on Windows and MacOs?

thomasdorch, 2022-11-01:
It doesn't work on Windows, I'm getting 'circular import' errors when trying to import gdspy.

Jan-David-Black, 2022-11-05:
I cannot get it to work on mac either :/
I would assume I am doing something wrong:

[image]

Selecting the python3.9 folder that comes with gdsfactory. When I run import_generic_PCells subsequently I get the following warnings/errors.

[image]
[image]

I installed `ubcpdk` in `~/gdsfactory/lib/python3.9/site-packages` but it doesn't make a difference.
Probbly I am doing something wrong. Let me know, if you have suggestions.

Jan-David-Black, 2022-11-05:
It seems that somehow the numpy integration wit KLayout got damaged in the process
[image]

thomasdorch, 2022-11-06:
Try uninstalling the `klayout` python package in your python environment. That should fix the `name gfc is not defined` error.

SkandanC, 2022-11-12:
[image]

Running into this error on Ubuntu 22 WSL. Not sure if something is wrong with one of the files, or if I did something wrong.

SkandanC, 2022-11-22:
Hi @joamatab,

I think most of the stuff on the checklist are completed by my latest PR, except for test on MacOS and test on Windows. We know it does not work on Windows, but idk if anyone has got it to work on MacOS yet. It wold be nice if someone with access to a MacOS computer can test it with the latest release.

Jan-David-Black, 2022-11-22:
I can do that. Could you tell me exactly what to test?
1. install new release
2. the try the gdsfactory menus in layout
or anything else?

SkandanC, 2022-11-22:
Thanks!

1, install new release
2. `gf tool install` in your terminal
3. Open KLayout -> gdsfactory 6.2.1 -> import generic PCells
4. Open new layout -> try importing random PCells from the gdsfactory library

Jan-David-Black, 2022-11-22:
release: 6.2.3
running `gf tool install` once everything seems fine
running it again gives:
```
NotADirectoryError: [Errno 20] Not a directory: '/Users/j.fischbach/.klayout/macros/gdsfactory'
```
`import generic PCells` prompts a file selector dialog

Jan-David-Black, 2022-11-22:
[image]

An I somehow have duplicate menu entries

Jan-David-Black, 2022-11-22:
After canceling the file dialog I now get
```
No module named 'flayout'
```

SkandanC, 2022-11-22:
I think the duplicate must be because the macro files must've been duplicated elsewhere, I will open a PR for this right away. They both should do the same thing.

The dialog box is expected - it wants you to select the directory where the `site-packages` directory sits in. This `site-packages` should also have gdsfactory and flayout in it. After the appropriate directory is selected, it should import the PCells automatically, displaying a progress bar.

Jan-David-Black, 2022-11-22:
I think I have selected the wrong path the first time. To reset it I have to delete the config json?

SkandanC, 2022-11-22:
Yes, it should be under ~/.gdsfactory

Jan-David-Black, 2022-11-22:
Hm it was installed...

Jan-David-Black, 2022-11-22:
Ah ok: I had the wrong directory: one level too deep...
now I get
```
No module named 'gdstk'
```
even though it is installed

Jan-David-Black, 2022-11-22:
sorry for the trouble

SkandanC, 2022-11-22:
> sorry for the trouble

No worries!

Jan-David-Black, 2022-11-22:
It seems that gdstk is not linked to klayout
I tried `lygadgets_link gdstk`
but it only gave me:
```
Did not recognize the klayout relevance of /User/<path to gdstk>
```

Jan-David-Black, 2022-11-22:
Ok now I only have the menu entries once: seemed to have been a double install also in salt.

SkandanC, 2022-11-22:
It must've been from your previous installation of the older version

Jan-David-Black, 2022-11-22:
So I cleaned everything
It still complains about gdstk missing

Jan-David-Black, 2022-11-22:
also in the macro ide:

[image]

SkandanC, 2022-11-22:
Are you able to import gdsfactory in python? If not, it must be an issue with the gdstk installation. If yes, then it must be an issue with KLayout's Python [similar to this](https://github.com/gdsfactory/gdsfactory/issues/891).

Jan-David-Black, 2022-11-22:
Yes I am able to run gdsfactory from my terminal

SkandanC, 2022-11-22:
Then it is the same issue as on Windows KLayout. I am glad we tested this, or else we could've overlooked this!

Jan-David-Black, 2022-11-22:
I might try a different gdstk install: instead of mamba directly pip. give me a sec

Jan-David-Black, 2022-11-23:
Ok, so tried pip and compile from source. Both didn't help...

joamatab, 2022-11-23:
I think you need to install gdsfactory on your MacOS python 

@lukasc-ubc

SkandanC, 2022-11-23:
I managed to install klayout-gui on Windows using @thomasdorch's conda install. I can import gdstk now 🎉:
[image]

But, it only is available on python3.11. The issue is gdsfactory does not support python3.11 because [one of our dependencies does not support python3.11](https://github.com/lxml/lxml/pull/356) https://github.com/lxml/lxml/pull/362 Once they make a new release, we can add support for python3.11, and then get gdsfactory on KLayout working on all 3 OS. @lukasc-ubc 

[image]

@Jan-David-Black can you try this on MacOS too, and see if gdstk can be imported there too?

Jan-David-Black, 2022-11-24:
> I think you need to install gdsfactory on your MacOS python

I seem to not fully understand: do you mean system level python? Should it not work with conda environments?

> @Jan-David-Black can you try this on MacOS too, and see if gdstk can be imported there too?
I will. (I'll edit as soon as it's done)

Due to relatively poor adaption of 3.11 so far I would prefer having the opportunity to stay with 3.10. Which will be possible as soon as @thomasdorch's build is up.

lukasc-ubc, 2022-11-28:
Can you folks help me get this running on Mac OSX?  Below is what I tried, and the error.    It would be helpful if there was a document that had the installation instructions.    Including which versions, which tools, etc.  

Thank you!
Lukas

* Installed v6.2.5 with the 1 GB installer.
* Installed in ~/gdsfactory
* ~/gdsfactory/bin/conda update GDSfactory
* Downloaded KLayout, “Anaconda3-environment based” LW 0.27.12
    * Copy and rename klayout to klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app
    * Copy and open the KLayoutAna3.app using Automator, and edit: export PYTHONHOME=/Users/lukasc/gdsfactory, and myklayout=/Applications/klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app
      * Can't save, error: The document “KLayoutAna3” could not be saved. A code signing error occurred.  /Applications/KLayoutAna3.app.sb-39bc65bd-FHY3tL: replacing existing signature
/Applications/KLayoutAna3.app.sb-39bc65bd-FHY3tL: unsealed contents present in the bundle root

    * Manually running in the terminal:
    * `export PYTHONHOME=/Users/lukasc/gdsfactory` 
    * `myklayout=/Applications/klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app`
    *  `myconfig=$HOME/.klayout/klayoutrc`
    * `open -n -a $myklayout --args -e -c $myconfig -style=fusion $@`
      * Errors: Termination Reason:    Namespace DYLD, Code 1 Library missing
      * Library not loaded: @rpath/libpython3.8.dylib
      * Referenced from: /Applications/klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app/Contents/MacOS/klayout
      * Reason: tried: '/Applications/klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app/Contents/MacOS/../Frameworks/libpython3.8.dylib' (no such file), '/Applications/klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app/Contents/MacOS/../Frameworks/libpython3.8.dylib' (no such file), '/Applications/klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app/Contents/MacOS/../Frameworks/libpython3.8.dylib' (no such file), '/Applications/anaconda3/lib/libpython3.8.dylib' (no such file), '/Applications/klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app/Contents/MacOS/../Frameworks/libpython3.8.dylib' (no such file), '/Applications/klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app/Contents/MacOS/../Frameworks/libpython3.8.dylib' (no such file), '/Applications/klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.app/Contents/MacOS/../Frameworks/libpython3.8.dyli

SkandanC, 2022-11-28:
Hi Prof. Chrostowski @lukasc-ubc,

By "Anconda3-environment based", do you mean the conda install by Thomas Dorch that I mentioned? Afaik, that is only meant for Windows as of now.

joamatab, 2022-12-01:
Hi Skandan, 

any changes I make on the Pcell I get a `NoneType` error

does it work well on your end?

SkandanC, 2022-12-01:
Hi @joamatab,

It works well on my end, could you post a screenshot, or the error message so I can debug it?

joamatab, 2022-12-05:
yes,

https://github.com/gdsfactory/gdsfactory/pull/957

whenever i change any parameter this is what happens

[image]

joamatab, 2023-09-12:
@nikosavola 

I think you had a way of making this work

closed 2023-11-12 by joamatab