gdsfactory/gdsfactory #802: create shortcuts during installation

joamatab, 2022-10-31:
How can we create shortcuts during installation?

we can use `menuinst` as suggested [here](https://github.com/conda/constructor/issues/573)

This continues the [simpler install](https://github.com/gdsfactory/gdsfactory/issues/723) effort

SkandanC, 2022-11-03:
@joamatab I'm just curious, do we know what the shortcut is supposed to do? As in, will it be used to open the docs upon clicking on it, or something else?

joamatab, 2022-11-03:
Hi Skandan, we can add shortcuts to spyder, anaconda navigator and jupyter lab

SkandanC, 2022-11-04:
[image]
After clicking on this icon:
[image]

I got Spyder shortcut installed :). I now have a template for adding shortcuts to all executables, so adding them for Anaconda Navigator and Jupyter lab should just be a matter of replacing "Spyder" in my script with these.

joamatab, 2022-11-04:
Great work Skandan :)

SkandanC, 2022-11-04:
Thanks Joaquin! Should we rename this icon as something like "Spyder with GDSFactory"?

joamatab, 2022-11-04:
yes, that would be great,

How about `gdsfactory spyder` `gdsfactory jupyterlab` and `gdsfactory navigator` for spyder, jupyterlab and anaconda navigator
?

joamatab, 2022-11-06:
Thank you Skandan!

there were some hard coded paths that i tried fixing in release  v5.54.0

let me know if that release creates the shortcuts correctly :)

SkandanC, 2022-11-06:
Woops, sorry about that, I must've double checked before sending it!

joamatab, 2022-11-29:
Thank you Skandan,

I will mark this as done

closed 2022-11-29 by joamatab