gdsfactory/gdsfactory #803: fix DEVSIM

joamatab, 2022-10-31:
Hi,

I was very happy when I learned that DEVSIM was integrated into gdsfactory (great work!). I updated it with the command:

`pip install gdsfactory --upgrade`

and then:

`pip install gdsfactory[full]`

I noticed that the DEVSIM package was installed:
[image]
However, the integration with gdsfactory doesn't works:
[image]
Also I tried to find some functions into gdsfactory.simulation related with DEVSIM:
[image]
and I didn't find any function. Then I deleted and installed gdsfactory (and plugins) and the issues persisted.

Is there something I did wrong? Is there any step in the update that I didn't do?

_Originally posted by @adophobr in https://github.com/gdsfactory/gdsfactory/discussions/799_

joamatab, 2022-10-31:
Hi Adolfo,

when you import gdsfactory, which version do you see?

I also see some issues with the current devsim plugin

[image]

@tcaduser
@simbilod

simbilod, 2022-10-31:
The MKL library was intentionally left out of the DEVSIM pip package, it needs to be installed separately

#pip install mkl# worked for me when testing standalone devsim

tcaduser, 2022-11-01:
The original poster was successfully loading devsim with the gdsfactory[full] package.  Their issue was with a gdsfactory module load.

I avoided making the Intel MKL a pip dependency, because Anaconda users would potentially have 2 different MKL installs in the same environment, totaling over a GB in memory.

devsim supports other lapack and blas libraries by setting the DEVSIM_MATH_LIBS environment variable.  The solver then switches from the Intel MKL Pardiso direct solver to SuperLU.  This may result in numerical differences in the results.  I will have an updated release soon, which will automatically search for libopenblas, liblapack, and then libblas if the Intel MKL is not installed.

simbilod, 2022-11-01:
the devsim subfolder seems to be missing a ___init___ file, I will fix

simbilod, 2022-11-01:
@joamatab fix pushed to #806

joamatab, 2022-11-01:
Great, thank you Simon,

just released v5.48.0 with the new fix

closed 2022-11-01 by joamatab

adophobr, 2022-11-01:
Thank you Simon,

it's working!

tcaduser, 2022-11-06:
With version 2.3.6 of devsim, it will search:
1. DEVSIM_MATH_LIBS environment variable
2. Intel MKL libraries
3. libopenblas.so:liblapack.so:libblas.so

It will automatically switch from MKL Pardiso to SuperLU direct solver if the MKL libraries are not available.

Please see:
https://github.com/devsim/devsim/blob/main/CHANGES.md