gdsfactory/gdsfactory #812: Meep material database does not have silicon at 1.3um

joamatab, 2022-11-03:
Meep material database does not have silicon refractive index at 1.3um

when running 
```
import gdsfactory.simulation.gmeep as gm
import gdsfactory.simulation as sim
import gdsfactory as gf
import meep as mp

wavelength_start = 1.26
wavelength_stop = 1.36
sim_settings = dict(wavelength_start=wavelength_start, wavelength_stop=wavelength_stop)

c = gf.components.mmi1x2()
gm.write_sparameters_meep(c, **sim_settings)
```

we get

```
ValueError: User specified frequency 0.7633587786259541 is above the Medium's limit, 0.7352941176470588.
```

@smartalech
@thomasdorch 
@SkandanC 
@simbilod

smartalecH, 2022-11-03:
Meep currently has four different kinds of Silicon models, none of which cover 1.3 $\mu$ m:

https://github.com/NanoComp/meep/blob/038a2e59866374aabe51d02d2a4de37a8045364c/python/materials.py#L1065-L1067

https://github.com/NanoComp/meep/blob/038a2e59866374aabe51d02d2a4de37a8045364c/python/materials.py#L13-L15

https://github.com/NanoComp/meep/blob/038a2e59866374aabe51d02d2a4de37a8045364c/python/materials.py#L38-L40

https://github.com/NanoComp/meep/blob/038a2e59866374aabe51d02d2a4de37a8045364c/python/materials.py#L55-L57

We can add another model, we just need a Lorentz-drude fit for it.

(I wouldn't call this a bug, but a feature enhancement).

joamatab, 2022-11-03:
Thank you Alec!

closed 2022-11-03 by joamatab