gdsfactory/gdsfactory #816: port issues when absorbing

joamatab, 2022-11-04:
This code produces some pins that don't rotate

```
import gdsfactory as gf

@gf.cell
def test():
    c = gf.Component()
    bend = c << gf.components.bend_circular()
    bend.rotate(-90)
    c.add_ports(bend.ports)
    c.absorb(bend)
    return c

if __name__ == "__main__":
    c = gf.Component("demo")
    bend = c << test()
    c.add_ports(bend.ports)
    c.show(show_ports=True)
```

[image]

tvt173, 2022-11-04:
is that the fault of `absorb()`? the ports are being added via `c.add_ports(bend.ports)` and not through absorb, right?

joamatab, 2022-11-04:
I think the issue is that the pins are defined as paths and not as polygons so they don't rotate

tvt173, 2022-11-04:
Ah, I see. I was confused. I thought you were talking about the actual port objects, but you're talking about pin shapes. That makes sense

SkandanC, 2022-11-06:
[image]
I was able to fix it. Final result after fix :)

joamatab, 2022-11-06:
Hi Skandan, 

yes, replacing the paths by polygons would fix the issue, ideally we keep them as rotated paths

SkandanC, 2022-11-06:
Got it, I will try to figure out a way to do that, I have some free time, I can work on this.

SkandanC, 2022-11-07:
Small update: I found that if I turn off all the layers except pinrec, I can see the pins be repeated twice:
[image]
One pair of them is in the correct orientation, which means the pins are being rotated as they should be, even thoguh they are FlexPaths.
However, the `absorb` method absorbs paths from the parent component, but it does not absorb the rotated paths for soem reason.

joamatab, 2022-11-07:
great, thank you Skandan!

closed 2022-11-07 by joamatab