gdsfactory/gdsfactory #828: Meep grating fdtd doesn't take etch depth as argument

SkandanC, 2022-11-08:
**Describe the bug**
Gmeep grating fdtd functions don't take etch depth as argument, although it is listed as one of the kwargs

**To Reproduce**
```py
import gdsfactory.simulation.gmeep as gm
gm.write_sparameters_grating(
    period=0.7, 
    fill_factor=0.5, 
    etch_depth=0.08,  # errors out here
    overwrite=True
)
```
```
/usr/local/lib/python3.7/site-packages/gdsfactory/simulation/gmeep/write_sparameters_grating.py in write_sparameters_grating(plot, plot_contour, animate, overwrite, dirpath, decay_by, verbosity, **settings)
    137         return np.load(filepath_npz)
    138 
--> 139     sim_dict = get_simulation_grating_fiber(**settings)
    140     sim = sim_dict["sim"]
    141     freqs = sim_dict["freqs"]

TypeError: get_simulation_grating_fiber() got an unexpected keyword argument 'etch_depth'
```
**Expected behavior**
The functions should modify the simgeometry as expected
**Suggested fix**
Either remove the argumetn altogether, or figure out a way to make this work.

joamatab, 2022-11-08:
great catch Skandan

closed 2022-11-08 by joamatab

SkandanC, 2022-11-08:
Could we keep the argument and just change the partof the code that computes the etch_depth?

We could have something like `etch_depth = etch_depth or (wg_thickness - slab_thickness)` here:

https://github.com/gdsfactory/gdsfactory/blob/06f893cfe7b194260501739637ca2a4490db4c1d/gdsfactory/simulation/gmeep/get_simulation_grating_fiber.py#L346

simbilod, 2022-11-08:
Etch depth is accounted for in the original version of this code (the one here might be out of date)

https://github.com/simbilod/optio/blob/master/optio/get_simulation_fiber.py

SkandanC, 2022-11-09:
Nevermind, looks like we can set the etch depth by setting the `slab_thickness` argument.

PR #829 (2022-11-08): fix grating coupler meep input parameters
fixes #828