gdsfactory/gdsfactory #842: Astar router not working when using avoid layers

joamatab, 2022-11-11:
How can we fix the Astar router when using avoid layers?

```
import gdsfactory as gf

c = gf.Component("get_route_astar_avoid_layers")
cross_section = gf.get_cross_section("metal1", width=3)
w = gf.components.straight(cross_section=cross_section)

left = c << w
right = c << w
right.move((100, 80))

obstacle = gf.components.rectangle(size=(100, 3), layer="WG")
obstacle1 = c << obstacle
obstacle2 = c << obstacle
obstacle1.ymin = 40
obstacle2.xmin = 25

port1 = left.ports["e2"]
port2 = right.ports["e2"]

routes = gf.routing.get_route_astar(
    component=c,
    port1=port1,
    port2=port2,
    cross_section=cross_section,
    resolution=5,
    distance=6.5,
    avoid_layers=("M1",),
)

c.add(routes.references)
c.show()
```

@SkandanC

joamatab, 2022-11-11:
Hi Skandan,

for some reason the a_star router is very slow when using `avoid_layers`

SkandanC, 2022-11-11:
[image]

Hello @joamatab,

Changing the resolution to 10 fixes this. I will look into why this is happening.