gdsfactory/gdsfactory #844: component.get_polygons() no longer returns vertices

simbilod, 2022-11-11:
component.get_polygons() no longer returns vertices

Example e.g.:

```
        c_layer = component.extract(layer["layer"])
        layer_boolean = boolean(c_layer, line, "and")
        polygons = layer_boolean.get_polygons()
        for polygon in polygons:
            # Min point is midpoint between two closest vertices to origin
            print(polygon)
```
prints `Polygon at layer 1, datatype 0, with 4 points` instead of the old numpy array, causing incompatibility with my gmsh plugin (and others I assume)

The docstring for get_polygons still reads
```
        Returns
            out : list of array-like[N][2] or dictionary
                List containing the coordinates of the vertices of each
                polygon, or dictionary with with the list of polygons (if
                `by_spec` is True).

```

We need to decide if we modify `get_polygons` to recover the old behaviour, or if we go forward assuming gdstk.polygon as the returned type

@joamatab

simbilod, 2022-11-11:
looks like by the time I complained it was fixed #843

simbilod, 2022-11-11:
Example of old return value for the same code:

```
[[  4.005  11.   ]
 [  3.995  11.   ]
 [  3.995 -11.   ]
 [  4.005 -11.   ]]
[[ 4.005  0.25 ]
 [ 3.995  0.25 ]
 [ 3.995 -0.25 ]
 [ 4.005 -0.25 ]]
[[4.005 9.35 ]
 [3.995 9.35 ]
 [3.995 8.65 ]
 [4.005 8.65 ]]
[[4.005 7.35 ]
 [3.995 7.35 ]
 [3.995 6.65 ]
 [4.005 6.65 ]]
[[4.005 5.35 ]
 [3.995 5.35 ]
 [3.995 4.65 ]
 [4.005 4.65 ]]
[[4.005 3.35 ]
 [3.995 3.35 ]
 [3.995 2.65 ]
 [4.005 2.65 ]]
[[ 4.005 -2.65 ]
 [ 3.995 -2.65 ]
 [ 3.995 -3.35 ]
 [ 4.005 -3.35 ]]
[[ 4.005 -4.65 ]
 [ 3.995 -4.65 ]
 [ 3.995 -5.35 ]
 [ 4.005 -5.35 ]]
[[ 4.005 -6.65 ]
 [ 3.995 -6.65 ]
 [ 3.995 -7.35 ]
 [ 4.005 -7.35 ]]
[[ 4.005 -8.65 ]
 [ 3.995 -8.65 ]
 [ 3.995 -9.35 ]
 [ 4.005 -9.35 ]]
```

SkandanC, 2022-11-11:
@joamatab This is the same issue causing the errors here https://github.com/gdsfactory/gdsfactory/issues/842

Fixing this would solve both these issues

tvt173, 2022-11-11:
I would vote for matching the old return value by default but adding an `as_array` argument (default True). when as_array=false, return the Polygon objects instead. polygon objects have more information (especially when by_spec=False) and will be faster to retrieve, so would be desirable to keep in some situations. but I agree it would be best to keep old behavior by default

closed 2022-11-11 by joamatab