gdsfactory/gdsfactory #848: meep plugin requires fixing for MPI

joamatab, 2022-11-11:
`pytest gdsfactory/simulation/gmeep/test_write_sparameters_meep.py` is not working with gdstk backend

the failing code is coming from
`write_sparameters_meep_mpi.py`

We can't pickle gdstk Cells so maybe it's cleaning writing and reading them using GDS instead of pickle

simbilod, 2022-11-11:
I'm on it

simbilod, 2022-11-11:
See #849

joamatab, 2022-11-11:
Thank you Simon for your PR, notice that the MPI tests are still failing

simbilod, 2022-11-11:
on my end it was failing due to h5py, so I thought it was something wrong with my setup

`ImportError: /usr/local/google/home/bilodeaus/miniconda3/envs/gdsfactory4_meep/lib/python3.10/site-packages/h5py/defs.cpython-310-x86_64-linux-gnu.so: undefined symbol: H5Pget_fapl_direct`

I can look into it more later

joamatab, 2022-11-11:
I get only wavelengths in the output

joamatab, 2022-11-12:
fixed, the ports were missing

`write_gds_with_metadata` instead of `write_gds` fixed the issue

closed 2022-11-12 by joamatab