gdsfactory/gdsfactory #852: Is there a way to get c.show() working on WSL?

SkandanC, 2022-11-12:
**Describe the bug**
Component.show() does not work on WSL/

**To Reproduce**
Running this:
```
import gdsfactory as gf
gf.components.mzi().show()
```
Produces no response, it's as though there is no KLayout window open.

**Expected behavior**
The KLayout window should open the MZI component

simbilod, 2022-11-12:
Where is your KLayout installed? WSL or Windows?

SkandanC, 2022-11-12:
Hi @simbilod ,

It's installed in Windows. Is there a way to install it on WSL?

simbilod, 2022-11-12:
You can install on WSL like you would any linux distribution, see https://klayout.de/build.html

Then when you run it (./klayout wherever you installed), it should open the window, and if you have latest WSL it should work like any other app https://learn.microsoft.com/en-us/windows/wsl/tutorials/gui-apps

Alternatively, there might be a way to setup the paths to go from WSL gdsfactory execution to Windows KLayout, but running KLayout from WSL seems like the easier option

SkandanC, 2022-11-12:
Ok that worked, ty @simbilod! Still ran into an issue when importing the cells. https://github.com/gdsfactory/gdsfactory/issues/800#issuecomment-1312556938

closed 2022-11-12 by SkandanC