gdsfactory/gdsfactory #868: difftest improvements

tvt173, 2022-11-14:
**Is your feature request related to a problem? Please describe.**
A few problems...
- when running difftest with `xor` on, I still sometimes see gdstk infinitely hang
- gdsfactory should not need to rely on phidl or gdspy anymore, but including `lytest` requires both... gdspy has installation issues on windows
- it is wasteful to re-run XORs in gdstk when we already run them once in klayout

**Describe the solution you'd like**
We should port `run_xor` out of `lytest` and give an option to export the results to a file. When `xor` is True in difftest, import and display the klayout-based results instead of re-running in gdstk. Then, we can drop lytest also.

**Suggested fix**
The above suggested solution should fix all three issues

joamatab, 2022-11-14:
Hi Troy,

I agree

here are two branches with some experiments in that direction

1. https://github.com/gdsfactory/gdsfactory/tree/604b

The issue with gdstk booleans is that we get spurious polygons that get flagged as a XOR diff

This branch adds klayout backend features
2. https://github.com/gdsfactory/gdsfactory/tree/klayout_backend_and_gdstk

tvt173, 2022-11-15:
i would _suspect_ that the spurious polygons _could_ be due to us acting on the in-memory shapes in gdstk, with full precision, rather than the polygons which have gone through GDS export and import which we are seeing in Klayout... would be interesting to try writing/reading the shapes and re-doing the booleans in gdstk to see if they get the same result. 

more seriously though, sometimes the XORs fail to finish and the function infinitely hangs. @flaport and I had seen this behavior in the past and @heitzmann helped to solve many of the issues, but it seems like it's not fully fixed. i don't fully remember the cause of these issues (I think it had to do with some strange corner cases with self-intersecting polygons, etc... @heitzmann, do you remember better or have an idea on what we should look into?)

joamatab, 2022-11-15:
Hi Troy, can you try after this [PR](https://github.com/gdsfactory/gdsfactory/pull/878)

closed 2022-11-15 by joamatab