gdsfactory/gdsfactory #877: How to fix font_free_type?

joamatab, 2022-11-15:
```
    from gdsfactory.components.text_freetype import text_freetype
    c2 = text_freetype(
        "hello",
        font="Times New Roman"
    )
    c2.show(show_ports=True)
```

Returns
```
File ~/gdsfactory/gdsfactory/component.py:137, in Component.__init__(self, name, with_uuid)
    134 if with_uuid or name == "Unnamed":
    135     name += f"_{self.uid}"
--> 137 self._cell = gdstk.Cell(name=name)
    138 self.name = name
    139 self.info: Dict[str, Any] = {}

ValueError: embedded null character

```

joamatab, 2022-11-15:
Try to fix it in here

https://github.com/gdsfactory/gdsfactory/pull/878

but still has some issues

JinWallner, 2022-12-01:
When I try to use text_freetype with ttf file.
    t = c << gf.components.text_freetype(text=text, size=size, layer=layer, font=currentdir+"/DEPLOF.ttf")
I got a different kind of error:
  File "/Users/jin/opt/anaconda3/lib/python3.9/site-packages/gdsfactory/components/text_freetype.py", line 89, in text_freetype
    letter_dev.add_polygon(poly.polygons, layer=layer)
AttributeError: 'gdstk.Polygon' object has no attribute 'polygons'

joamatab, 2022-12-02:
Hi Jin, this needs fix since we migrated to gdstk 

@SkandanC 
@tvt173

wuzhonghan, 2023-03-03:
this issue still there

joamatab, 2023-03-03:
yes, we are looking forward to having someone help fixing this

SkandanC, 2023-03-16:
[image]

Hello @JinWallner!

I will open a PR for this!

closed 2023-03-16 by joamatab