gdsfactory/gdsfactory #886: `gf tool install` does not create "gdsfactory tab" on KLayout menu bar

SkandanC, 2022-11-16:
**Describe the bug**
According to the docs, running `gf tool install` should install the KLayout salt package, and must be similar to installing it straight from KLayout. This should create a tab on the menu bar to import PCells.

**To Reproduce**
Run `gf tool install` in terminal.
[image]

**Suggested fix**
Have gf tool install install the .lym files under `macros/gdsfactory` rather than `salt/gdsfactory`. Doing so produces this:
[image]

I will open a PR soon with this fix.

SkandanC, 2022-11-16:
Turns out the issue is pip install gdsfactory does not install any of the klayout macro files except for klive.lym (this has been the case for me on Windows and WSL Ubuntu). I am not sure why this is the case. Perhaps @thomasdorch does?

Conda installed gdsfactory's command line tool is broken:
[image]

thomasdorch, 2022-11-16:
Hmm, I think I have run into the first issue you mentioned before, but I am not able to recreate it now..

On my system (Win11), I am able to run `pip install -e .` to install gdsfactory, then run `gf tool install` and the menu options show up on the KLayout toolbar, regardless if it uses a symlink or junction.. What command are you running to pip install gdsfactory?

Also, I think we should keep the gdsfactory files all in the same place (salt) if possible, rather than splitting them up between the different folders.

SkandanC, 2022-11-16:
> What command are you running to pip install gdsfactory?

Just `pip install gdsfactory`. I've tried this in multiple conda envs with different python versions, and on both Windows and WSL - same issue in all.

> Also, I think we should keep the gdsfactory files all in the same place (salt) if possible, rather than splitting them up between the different folders.

Yep, I don't think there is a need to move them to another place now, if we can fix the above issue.

thomasdorch, 2022-11-16:
Okay it looks like it's because the rest of the macros aren't included in the MANIFEST.in. So if we replace 

```
include gdsfactory/klayout/pymacros/klive.lym
```
with
```
include gdsfactory/klayout/pymacros/*
```
I think it should fix this.

thomasdorch, 2022-11-16:
I also ran into the same errors with conda-installed gdsfactory not working..

commit f1cd8ab8 (2022-11-17): Include all pymacros when building package

(hopefully) fixes #886
commit 56f5450e (2023-02-20): Include all pymacros when building package

(hopefully) fixes #886

Former-commit-id: f1cd8ab86498a2e459c52c5c5d127ffa5e826d39
commit cab3da5d (2023-02-20): Include all pymacros when building package

(hopefully) fixes #886

Former-commit-id: 5cb7183a5c4c0177d22fd141d716e07b7acae038 [formerly f1cd8ab86498a2e459c52c5c5d127ffa5e826d39]
Former-commit-id: 56f5450e4f46e32b280d1eebfdbea9da79d2e3a2
commit f7a8d5e1 (2023-02-23): Include all pymacros when building package

(hopefully) fixes #886

Former-commit-id: 5cb7183a5c4c0177d22fd141d716e07b7acae038 [formerly 41d2f17f71daab7f16b70e133abf6e585104a0f4 [formerly 9612671424ce48d8d2518ab936e6015079037d3e] [formerly 56f5450e4f46e32b280d1eebfdbea9da79d2e3a2 [formerly f1cd8ab86498a2e459c52c5c5d127ffa5e826d39]]]
Former-commit-id: 56f5450e4f46e32b280d1eebfdbea9da79d2e3a2
Former-commit-id: 6b0e65ddd3c31c587af911fac5080f7da36e7777 [formerly cab3da5dab0aeb6007a07d779da6aa2de96a8826]
Former-commit-id: 84f40600fa02b0015b3c79245cb79d18ad530042