gdsfactory/gdsfactory #888: gf.geometry.outline(.. open_ports=True...) returning odd error

bobnifty, 2022-11-17:
**Describe the bug**
gf.geometry.outline(open_ports=True)
gives an error:
"ValueError: port = 'E' not in 'compass_size10p0006__20p0' ports ['e1', 'e2', 'e3', 'e4']"

**To Reproduce**
comp1 = gf.components.taper(length=100,width1=20,width2=50)
outline_comp1 = gf.geometry.outline(comp1,open_ports=True)

**Expected behavior**
Returns outline of original component with the ports 'open' (not outlined)

**Suggested fix**
Maybe reverting to the phidl wrap?

bobnifty, 2022-11-17:
Actually it seems like just pointing the port to 'e3' seems to work....

joamatab, 2022-11-19:
Thank you Robert for posting the issue and suggesting the fix

https://github.com/gdsfactory/gdsfactory/pull/893/files

releasing 6.1.1 with the fix and a test for it

closed 2022-11-19 by joamatab