gdsfactory/gdsfactory #889: clean_value_json should handle Polygons

tvt173, 2022-11-17:
Sometimes it is desirable to pass a Polygon, list of Polygon, or list of numpy arrays representing polygons as an argument, but it currently breaks `clean_value_json`. Would be great if that worked.

joamatab, 2022-11-17:
Hi Troy, can you add an example for the code to reproduce?

tvt173, 2022-11-18:
hi @joamatab , please see the script below

```python
import gdsfactory as gf

@gf.cell
def wrap_polygon(polygon) -> gf.Component:
    c = gf.Component()
    # c.add_polygon(polygon, layer="WG")
    return c

@gf.cell
def wrap_polygons(polygons) -> gf.Component:
    c = gf.Component()
    # c.add_polygon(polygons, layer="WG")
    return c

if __name__ == '__main__':
    import gdstk

    # Try individually toggling on/off the commented lines below to see failing/passing cases

    # c = wrap_polygon(gdstk.rectangle((0,0), (1,1)))  # FAILS

    s = gf.components.straight()
    # c = wrap_polygons(s.get_polygons(as_array=False))  # FAILS
    # c = wrap_polygons(s.get_polygons(by_spec=False, as_array=True))  # WORKS
    c = wrap_polygons(s.get_polygons(by_spec=True, as_array=True))  # WORKS

    s = gf.components.ring_double_heater()
    # c = wrap_polygons(s.get_polygons(by_spec=False, as_array=True))  # FAILS
    # c = wrap_polygons(s.get_polygons(by_spec=(1, 0), as_array=True))  # FAILS
    # c = wrap_polygons(s.get_polygons(by_spec=True, as_array=True))  # FAILS

    c.show()
```

joamatab, 2022-11-19:
Thank you Troy, just released 6.1.1

closed 2022-11-19 by joamatab