gdsfactory/gdsfactory #891: KLayout on Windows struggles to import gdstk while running gf macros

SkandanC, 2022-11-17:
**Describe the bug**
When trying to import PCell library in KLayout, it fails to import gdstk, and the entire process doesn't work.

**To Reproduce**
gdsfactory menu -> import_generic_PCells. I modified the error message only to make it more verbose.
[image]

**Suggested fix**
This is a known issue with KLayout's Python. 
https://www.klayout.de/forum/discussion/1834/nazca-module-is-not-recognized

SkandanC, 2022-11-17:
It can import flayout perfectly fine, it just can't import gdstk because it is stored as a .pyd under the same directory. KLayout's Python cannot import .pyd

thomasdorch, 2022-11-17:
Yeah this is the same issue I had with gdspy before :/

tvt173, 2022-11-18:
have you tried setting the PYTHONPATH to the directory with the *.pyd files?
https://stackoverflow.com/questions/50278029/how-to-import-a-pyd-file-as-a-python-module

SkandanC, 2022-11-18:
Hey @tvt173,

According to Matthias in the link I posted, this won't work.
[image]

thomasdorch, 2022-11-18:
My understanding is that the KLayout Windows program is compiled with MSYS2 while Python C-extensions are all compiled with the MSVS compilers, hence no binary-compatibility.

tvt173, 2022-11-21:
> My understanding is that the KLayout Windows program is compiled with MSYS2 while Python C-extensions are all compiled with the MSVS compilers, hence no binary-compatibility.

does that mean you could potentially fix the issue by compiling Klayout with the MSVS compiler?

better long term solution might be to launch a gdsfactory server on a separate process (choose which python you like) and communicate with Klayout via a client API.

thomasdorch, 2022-11-21:
> > My understanding is that the KLayout Windows program is compiled with MSYS2 while Python C-extensions are all compiled with the MSVS compilers, hence no binary-compatibility.
> 
> does that mean you could potentially fix the issue by compiling Klayout with the MSVS compiler?

Absolutely! I've been able to get the [GUI version built on Windows](https://github.com/thomasdorch/klayout-conda) using the MSVC2017 compiler and it works just fine with gdstk :)

tvt173, 2022-11-21:
@thomasdorch I think a conda install is a great option. nice work on getting that running! but it would also be great for the standard Klayout to have better support for these packages as well. I'd imagine this would be an issue for many compiled python packages and not just the ones we're dealing with here, right?

@klayoutmatthias, is there any particular reason for choosing MSYS2 as the compiler for Klayout Windows version, or would you be open to changing to get better python support?

tvt173, 2022-11-21:
thanks Matthias. sounds good

closed 2023-09-12 by joamatab