gdsfactory/gdsfactory #910: The straight=<function straight> parameter in get_bundle is not active

JinWallner, 2022-11-22:
The straight parameter in get_bundle is not active. It seems not to get passed to the function calls inside get_bundle. It's working in get_route.

joamatab, 2022-11-23:
Hi Jin,

Why don't you use the `cross_section` argument for `get_bundle`?

How can we reproduce your issue?

This works well

```
    c = gf.Component("demo")
    c1 = c << gf.components.mmi2x2()
    c2 = c << gf.components.mmi2x2()
    c2.move((100, 40))
    routes = get_bundle(
        [c1.ports["o2"], c1.ports["o1"]],
        [c2.ports["o1"], c2.ports["o2"]],
        radius=5,
        layer=(2, 0),
    )
    for route in routes:
        c.add(route.references)
    c.show(show_ports=True)
```

joamatab, 2022-11-23:
Now we also allow to use custom straight functions in get_bundle

```
    c = gf.Component("demo")
    c1 = c << gf.components.mmi2x2()
    c2 = c << gf.components.mmi2x2()
    c2.move((100, 40))
    routes = get_bundle(
        [c1.ports["o2"], c1.ports["o1"]],
        [c2.ports["o1"], c2.ports["o2"]],
        radius=5,
        # layer=(2, 0),
        straight=gf.partial(gf.components.straight, layer=(1, 0), width=1),
    )
    for route in routes:
        c.add(route.references)
    c.show(show_ports=True)
```

closed 2022-11-23 by joamatab

JinWallner, 2022-11-23:
Hi Joaquin,

I was trying to print INTERCONNECT model information in the routing
components so I can use the Lumerical-Klayout package to extract the
INTERCONNECT netlist from the GDS. I modified your example with added
straight and bend components. From the output gds, you can see that the
bend has the model information, but the straight doesn't have it.

Thanks,

Jin

import gdsfactory as gf
from gdsfactory.add_labels import add_siepic_labels

strip_wg_simulation_info = dict(
model="wg",
layout_model_property_pairs=(
("length","wg_length"),
("width","wg_width")
),
layout_model_port_pairs=(("o1","port 1"), ("o2","port 2")),
properties=dict(annotate=False),
)

my_add_label = gf.partial(add_siepic_labels,model="o_strip",library="abc",
spice_params="auto", label_spacing=0.1)
mystraight = gf.partial(gf.components.straight, cross_section="strip",info=
strip_wg_simulation_info, decorator=my_add_label)
mybend = gf.partial(gf.components.bend_euler, cross_section="strip",angle=90
,p=0.26,info=strip_wg_simulation_info, decorator=my_add_label)

c = gf.Component("Test")
c1 = c << gf.components.mmi2x2()
c2 = c << gf.components.mmi2x2()
c2.move((100, 40))
routes = gf.routing.get_bundle(
[c1.ports["o2"], c1.ports["o1"]],
[c2.ports["o1"], c2.ports["o2"]],
radius=5,
cross_section='strip',
bend=mybend,
straight=mystraight
)
for route in routes:
c.add(route.references)
c.show(show_ports=True)

joamatab, 2022-11-23:
Hi Jin, 

layout properties are now propagating to the route references

[image]

so if you update to `6.2.4` it should work for you

@thomasdorch  wrote the interconnect plugin

JinWallner, 2022-11-28:
It's working well now. Thank you. -Jin

PR #914 (2022-11-23): straight propagates to route filter in get_bundle
Now we also allow to use custom straight functions in get_bundle

```
    c = gf.Component("demo")
    c1 = c << gf.components.mmi2x2()
    c2 = c << gf.components.mmi2x2()
    c2.move((100, 40))
    routes = get_bundle(
        [c1.ports["o2"], c1.ports["o1"]],
        [c2.ports["o1"], c2.ports["o2"]],
        radius=5,
        # layer=(2, 0),
        straight=gf.partial(gf.components.straight, layer=(1, 0), width=1),
    )
    for route in routes:
        c.add(route.references)
    c.show(show_ports=True)
```

fixes #910