gdsfactory/gdsfactory #922: The matplotlib drawing window cannot display images and automatically closed at a very fast speed.

HWOde, 2022-11-29:
For example, the following code

import gdsfactory as gf
@gf.cell
def mzi_with_bend(radius: float = 10):
    c = gf.Component()
    mzi = c << gf.components.mzi()  # equivalent to mzi = c.add_ref(gf.components.mzi())
    bend = c << gf.components.bend_circular(
        radius=radius
    )  # equivalent to bend = c.add_ref(gf.components.bend_circular())
    bend.connect("o1", mzi.ports["o2"])
    return c

c = mzi_with_bend(radius=20)
c.plot()
c.show()

There will be corresponding structure output in the klayout software, but the following images cannot be displayed in the matplotlib drawing window. 
[image]

My Python interpreter version：python 3.9.13
                   gdsfactory version:   6.2.6
                   matplotlib version：3.5.2
                    Operating system：Windows11 
                                            IDE: Pycharm

SkandanC, 2022-11-29:
Hi @HWOde,
did you try adding a `plt.show()` line after `c.plot()`? That is what I do if I want the window to stay open.

HWOde, 2022-11-29:
Bravo, it works!😊@ SkandanC

closed 2022-11-29 by HWOde

HWOde, 2022-11-29:
@SkandanC  😊 Thank you so much for your kind help and have a good day！