gdsfactory/gdsfactory #934: Layout versus schematic LVS

joamatab, 2022-11-29:
As circuits become more complex is nice to keep a design intent that you can check against the produced layout

[image]

https://github.com/gdsfactory/gdsfactory/issues/187
https://github.com/gdsfactory/gdsfactory/issues/931

We just added [schematic driven layout](https://gdsfactory.github.io/gdsfactory/notebooks/20_schematic_driven_layout.html)  and we have an MVP for netlist extraction

How can we use netgen or other tools developed in electronics? in the end we just need to match 2 connectivity graphs

das-dias, 2022-11-30:
> As circuits become more complex is important to check that your design
> intent matches the produced layout
> @das-dias <https://github.com/das-dias>
> @proppy <https://github.com/proppy>
> @tvt173 <https://github.com/tvt173>
> @sebastian-goeldi <https://github.com/sebastian-goeldi>
> @damienbonneau <https://github.com/damienbonneau>

Hello Joaquin,

I have never written software for LVS extraction, but my experience tells
me there are two main ways to tackle this problem.

So, Layout-Versus-Schematic is an operation in which the netlist generated
from the schematic is compared against the netlist generated from the
layout. This comparison works by checking if the same number of nets and
devices are comprised within the netlist. Thinking in terms of graphs, if
every electronic device's ports, input and output ports, as well as VDD and
VSS, are the graph's nodes, then the nets will be the transitions between
those nodes, and therefore, the graphs generated from schematic and layout
simply have to be topologically equivalent. VDD and VSS are special nodes
of the graph, because in my analogy they can be a node and a transition
simultaneously.

The first way you can go about this is by encoding which nets belong to
each interconnect layer within the GDS information of every PCELL existing
in a given technology. For a PMOS, for example, you create a premade layout
cell instance of the device, in which the source, drain and gate/back-gate
connections are already designed and labelled with the corresponding nodes
that will be featured in a netlist. If you connect a labelled polygon
representing VDD to the source of the PMOS PCELL in the layout, then when
comparing it against the schematic you know you will be able to see the
same connection between the source of the PMOS device to VDD in the
schematic. Of course, the names have to match, and usually, this means that
some names are globally set and immutable - like !vdd, !vss, !source, etc.
These LVS rules can be thought of as predefined within the technology
rule's file, and can help to speed up as well as increasing the accuracy of
LVS reports.

The other way you could go about this is by ignoring the names, and simply
inferring the number of nodes and transitions of both graphs. If both
graphs present the same number of nodes and transitions, this will probably
mean that the layout matches the schematic, so there's a confidence level
through which you can do this. This is a more general way of doing LVS, and
can be applied to a broader spectrum of layout topologies.

Technology nodes usually feature a mix of both approaches to tackle this
problem, offering good flexibility, speed and accuracy simultaneously.

I hope I had some useful insight to offer.

Best regards,

Diogo André

à(s) 19:57:

tvt173, 2022-11-30:
One good way to do an LVS check would be to convert the netlist into a graph, with networkx, and use network's built-in `is_isomorphic()` or (more advanced) `GraphMatcher` functionality. Please see initial proof-of-concept below:
https://networkx.org/documentation/stable/reference/algorithms/isomorphism.html

```python
import networkx

instances = {
    'inst1': 'waveguide',
    'inst2': 'waveguide',
    'inst3': 'detector',
}

component_ports = {
    'waveguide': ['p1', 'p2'],
    'detector': ['p1'],
}

nets1 = [
    ['inst1,p2', 'inst2,p1'],
    ['inst2,p2', 'inst3,p1'],
]

nets2 = [
    ['inst2,p2', 'inst1,p1'],
    ['inst1,p2', 'inst3,p1'],
]

def nodes_exactly_equal(node1: dict, node2: dict):
    return node1 == node2

def components_match(node1: dict, node2: dict):
    return node1['component'] == node2['component']

def component_and_port_matches(node1: dict, node2: dict):
    keys = ['component', 'port']
    for key in keys:
        if node1[key] != node2[key]:
            return False
    return True

if __name__ == '__main__':

    g1 = networkx.Graph()
    g2 = networkx.Graph()

    for inst, component in instances.items():
        ports = component_ports[component]
        for port in ports:
            full_port_name = f'{inst},{port}'
            g1.add_node(full_port_name, component=component, port=port, inst=inst)
            g2.add_node(full_port_name, component=component, port=port, inst=inst)

    for net in nets1:
        g1.add_edge(*net)

    for net in nets2:
        g2.add_edge(*net)

    print('Netlists are isomorphic (no node check):', networkx.is_isomorphic(g1, g2, node_match=None))
    print('Netlists exactly match:', networkx.is_isomorphic(g1, g2, node_match=nodes_exactly_equal))
    print('Components match:', networkx.is_isomorphic(g1, g2, node_match=components_match))
    print('Component and ports match:', networkx.is_isomorphic(g1, g2, node_match=component_and_port_matches))
```

Outputs:
```console
Netlists are isomorphic (no node check): True
Netlists exactly match: False
Components match: True
Component and ports match: True
```

proppy, 2022-11-30:
@d-m-bailey from https://github.com/RTimothyEdwards/netgen fame might also have some guidance on this (and they're  also based in Japan!)

d-m-bailey, 2022-12-01:
A couple considerations when using graph matching for LVS:
1. Some devices have terminals that are interchangeable. For example, resistors or the source and drain of many mosfets. You'd need some way to match netlist 1 with a mosfet source=VDD, drain=OUT to netlist2 with a mosfet  source=OUT, drain=VDD.
2. Often devices in the schematic will be simplified versions of those in the layout. For example, the layout could have 10  mosfets with W=10, L=1, but the schematic would only have 1 with W=100, L=1. These need to be reduced and matched as equivalent.
3. Even though the topology is the same, sometimes the terminals names have been placed incorrectly. After the topology match, a port match check (with automorphisms) is probably necessary.
4. Parameter checking usually occurs after graph matching  but maybe necessary to resolve automorphisms (circuits with identical connectivity graphs).

joamatab, 2022-12-05:
Here are some nice examples on DRC and LVS

https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr

das-dias, 2023-01-19:
I have been exploring this issue a little bit more and I think KLayout is the way to go. From what I've seen in the repo that Joaquin showed 

> https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_pr

I think that KLayout offers two ways of performing LVS:
- either by using KLayout lib calls to its functions
- or by evoking klayout in the terminal to perform LVS and generate a report.

The question here is: do we want to create gdsfactory's own LVS tool inspired on, and even using, klayout and netgen functions and methods or do we want to simply integrate a python script that performs an OS call to klayout in the terminal to obtain the LVS output?

das-dias, 2023-01-20:
Responding to Mitch Bailey: I think those issues can be handled by collapsing multiple identified connections of the same two nodes into a single instance of the node, in both the netlist extracted from the spice file as well as the netlist extracted from layout. Multiple devices in parallel to implement a transistor width multiplier (by shorting their gates, sources, drains and bulks) can eventually get collapsed into a single device during the construction of the connectivity graph - by checking if all of their terminals are shorted in the layout connectivity graph for example. But by lack of experience in such programming activities, I actually don't know how straightforward this is to implement. Even so, I agree that reutilising KLayout and netgen's capabilities is the best option, once these tools are already matured in these kinds of technical issues. Let me know what you think.

d-m-bailey, 2023-01-20:
@das-dias klayout does hierarchical extraction, I believe, and may not be able to recognize (i.e. correctly extract) devices resulting from shapes in different hierarchies. Commercial tools do automatic flattening of cells to extract correctly. magic may also have difficulties (i.e. not extract correctly) with some non-standard cell layouts. magic and netgen currently are probably not suited for large designs. I've been running magic extraction on a sky130 caravel layout with 8 sram instances for the past 4 days. netgen uses a fixed hash size of around 50000, so any cell (or flattened cell) with more that that number of instances is going to see degradation in performance. The algorithm for flattening unmatched cells to see if they match at a higher level is also inefficient.

das-dias, 2023-01-20:
@d-m-bailey if by flattening you mean collapsing all the hierarchical levels of all the cells in the design to a single hierarchical level, I can see how the performance might suffer from severe degradation.  Plus, another thing that raises concerns for me is that it is not practical nor is it designer-friendly. If larger designs mean that the design flow will severely slow down, then I think an alternative that is enabled to perform an LVS comparison independently of the hierarchical level should be explored. But this means reinventing parts of the wheel I think.

SkandanC, 2023-04-01:
Hi,

KFactory has a Netlist extractor which uses KLayout API.

Example netlist extracted for an MZI.
```python
from kfactory.pcells.mzi import mzi

mzi1 = mzi()
mzi2 = mzi(delta_length=5)
c = kf.KCell()
mzi1 = c << mzi1
mzi2 = c << mzi2

mzi2.connect("o1", mzi1.ports["o3"])
c.add_port(port=mzi1.ports["o1"], name="o1")
c.add_port(port=mzi1.ports["o2"], name="o2")
c.add_port(port=mzi2.ports["o3"], name="o3")
c.add_port(port=mzi2.ports["o4"], name="o4")
c.draw_ports()
c.write("mzi.gds")
l2n = kf.kdb.LayoutToNetlist(c.begin_shapes_rec(0))
for l_idx in c.klib.layer_indices():
    l2n.connect(l2n.make_layer(l_idx, f"layer{l_idx}"))
l2n.extract_netlist()
print(l2n.netlist().to_s())
```

```circuit waveguide_W1000_L10000_L0_E8a9ff7cf ($1=$1);
end;
circuit straight_coupler_G0p2_L10_W1_L0_E8a9ff7cf ($1=$I2,$2=$I1);
  subcircuit waveguide_W1000_L10000_L0_E8a9ff7cf $1 ($1=$I2);   
  subcircuit waveguide_W1000_L10000_L0_E8a9ff7cf $2 ($1=$I1);   
end;
circuit bend_s_W1_H1p9_L5_L0_NP99_TS0_TS1_E8a9ff7cf ($1=$1);    
end;
circuit waveguide_W1000_L4500_L0_ENone ($1=$1);
end;
circuit waveguide_W1000_L160_L0_ENone ($1=$1);
end;
circuit waveguide_W1000_L2000_L0_ENone ($1=$1);
end;
circuit waveguide_W1000_L100_L0_ENone ($1=$1);
end;
circuit waveguide_W1000_L7000_L0_ENone ($1=$1);
end;
circuit bend_euler_W1_R5_L0_ENone_T90_R150 ($1=$1);
end;
circuit coupler_G0p2_L10_D5_D5_W1_L0_ENone ($1=$I4,$2=$I1);     
  subcircuit bend_s_W1_H1p9_L5_L0_NP99_TS0_TS1_E8a9ff7cf $1 ($1=$I4);
  subcircuit straight_coupler_G0p2_L10_W1_L0_E8a9ff7cf $2 ($1=$I4,$2=$I1);
  subcircuit bend_s_W1_H1p9_L5_L0_NP99_TS0_TS1_E8a9ff7cf $3 ($1=$I4);
  subcircuit bend_s_W1_H1p9_L5_L0_NP99_TS0_TS1_E8a9ff7cf $4 ($1=$I1);
  subcircuit bend_s_W1_H1p9_L5_L0_NP99_TS0_TS1_E8a9ff7cf $5 ($1=$I1);
end;
circuit mzi_DL5_LY2_LX0p1_BCbend_euler_Swaveguide_SYNone_SXTNone_SXBNone_Scoupler_CNone_WSTrue_PESo3_PESo4_PECo2_PECo1_N2_W1_L0_R5_ENone_K_empty ($1=$I9,$2=$I2);
  subcircuit coupler_G0p2_L10_D5_D5_W1_L0_ENone $1 ($1=$I2,$2=$I9);
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $2 ($1=$I9);    
  subcircuit waveguide_W1000_L4500_L0_ENone $3 ($1=$I9);        
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $4 ($1=$I9);    
  subcircuit waveguide_W1000_L100_L0_ENone $5 ($1=$I9);
  subcircuit waveguide_W1000_L160_L0_ENone $6 ($1=$I9);
  subcircuit coupler_G0p2_L10_D5_D5_W1_L0_ENone $7 ($1=$I2,$2=$I9);
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $8 ($1=$I9);    
  subcircuit waveguide_W1000_L4500_L0_ENone $9 ($1=$I9);        
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $10 ($1=$I9);   
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $11 ($1=$I2);   
  subcircuit waveguide_W1000_L2000_L0_ENone $12 ($1=$I2);       
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $13 ($1=$I2);   
  subcircuit waveguide_W1000_L100_L0_ENone $14 ($1=$I2);        
  subcircuit waveguide_W1000_L160_L0_ENone $15 ($1=$I2);        
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $16 ($1=$I2);   
  subcircuit waveguide_W1000_L2000_L0_ENone $17 ($1=$I2);       
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $18 ($1=$I2);   
end;
circuit mzi_DL10_LY2_LX0p1_BCbend_euler_Swaveguide_SYNone_SXTNone_SXBNone_Scoupler_CNone_WSTrue_PESo3_PESo4_PECo2_PECo1_N2_W1_L0_R5_ENone_K_empty ($1=$I11,$2=$I1);
  subcircuit coupler_G0p2_L10_D5_D5_W1_L0_ENone $1 ($1=$I11,$2=$I1);
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $2 ($1=$I11);   
  subcircuit waveguide_W1000_L2000_L0_ENone $3 ($1=$I11);       
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $4 ($1=$I11);   
  subcircuit waveguide_W1000_L100_L0_ENone $5 ($1=$I11);        
  subcircuit waveguide_W1000_L160_L0_ENone $6 ($1=$I11);        
  subcircuit coupler_G0p2_L10_D5_D5_W1_L0_ENone $7 ($1=$I11,$2=$I1);
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $8 ($1=$I11);   
  subcircuit waveguide_W1000_L2000_L0_ENone $9 ($1=$I11);       
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $10 ($1=$I11);  
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $11 ($1=$I1);   
  subcircuit waveguide_W1000_L7000_L0_ENone $12 ($1=$I1);       
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $13 ($1=$I1);   
  subcircuit waveguide_W1000_L100_L0_ENone $14 ($1=$I1);        
  subcircuit waveguide_W1000_L160_L0_ENone $15 ($1=$I1);        
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $16 ($1=$I1);   
  subcircuit waveguide_W1000_L7000_L0_ENone $17 ($1=$I1);       
  subcircuit bend_euler_W1_R5_L0_ENone_T90_R150 $18 ($1=$I1);   
end;
circuit Unnamed_12 ();
  subcircuit mzi_DL10_LY2_LX0p1_BCbend_euler_Swaveguide_SYNone_SXTNone_SXBNone_Scoupler_CNone_WSTrue_PESo3_PESo4_PECo2_PECo1_N2_W1_L0_R5_ENone_K_empty $1 ($1='o2,o4',$2='o1,o3');
  subcircuit mzi_DL5_LY2_LX0p1_BCbend_euler_Swaveguide_SYNone_SXTNone_SXBNone_Scoupler_CNone_WSTrue_PESo3_PESo4_PECo2_PECo1_N2_W1_L0_R5_ENone_K_empty $2 ($1='o2,o4',$2='o1,o3');
end;
```

sebastian-goeldi, 2023-04-01:
Well, yeah, buuut. What happens in case you have a directional coupler? ;)

SkandanC, 2023-04-01:
Hi @sebastian-goeldi,

This example does have a directional coupler, `coupler_G0p2_L10_D5_D5_W1_L0_ENone`

sebastian-goeldi, 2023-04-01:
Fair, but therein also lies the problem a bit.

Did you try to visualize this as a schema? Afaik this will create you a circuit that is completely parallel, e.g. if you chain 10 waveguides, I think this will generate you 10 parallel waveguides in your netlist, which is wrong

sebastian-goeldi, 2023-04-01:
Input and output should not end up on the same net. Each photonic component acts (at the very least) like a resistor in electronics.

What you can do to fix it is use ports as a way to emulate this. I.e. the ports are the contacts and the waveguide is the resistor.

SkandanC, 2023-04-02:
I did try to visualize it as a schema, and unless I'm not nterpreting this right, this netlist describes the different nets in each component, and which net of each of its components belong in which net, which gives an idea of how the different components are connected.

This might eb a trivial question, but is just checking for connectvities sufficient for photonic components? Do we not have to check for things like path length differences, and radius of curvature, etc.? Or do we already have these implemented in the format we plan to use (yaml, etc.)? Afaik, spice format and other formats in lvs for electronics cant be used here if we plan on using these things too.

sebastian-goeldi, 2023-04-02:
The problem with your approach is:

Let's say you have a waveguide, a y-splitter and two waveguides connected to each output. With this approach you cannot determine the order, which basically renders LVS useless because we are interested what is connected but also how

SkandanC, 2023-04-02:
Ah I see, understood. So I think whicheber format or netlist extractor we use should have to take into account the connections (how and what is connected to what), and also the device parameters such as radius of curvature.

joamatab, 2023-04-02:
This is amazing 🤩 

Maybe we can start with electrical LVS

How do we define the connection stack programmatically?

SkandanC, 2023-04-02:
Sweet.

We can also write to spice directly.

```from kfactory.pcells.mzi import mzi

from kfactory.simulation.write_sparameters_lumerical import plot_sparameters_lumerical

mzi1 = mzi()
mzi2 = mzi(delta_length=5)
c = kf.KCell()
mzi1 = c << mzi1
mzi2 = c << mzi2

mzi2.connect("o1", mzi1.ports["o3"])
c.add_port(port=mzi1.ports["o1"], name="o1")
c.add_port(port=mzi1.ports["o2"], name="o2")
c.add_port(port=mzi2.ports["o3"], name="o3")
c.add_port(port=mzi2.ports["o4"], name="o4")
c.draw_ports()
c.write("mzi.gds")
import kfactory as kf
l2n = kf.kdb.LayoutToNetlist(c.begin_shapes_rec(0))
for l_idx in c.klib.layer_indices():
    l2n.connect(l2n.make_layer(l_idx, f"layer{l_idx}"))
l2n.extract_netlist()
print(l2n.netlist().to_s())
l2n.netlist().write("mzi.cir", kf.kdb.NetlistSpiceWriter())
```

```

* cell Unnamed_12
.SUBCKT Unnamed_12
* net 1 o1,o3
* net 2 o2,o4
* cell instance $1 r0 *1 0,0
X$1 2 1
+ mzi_DL10_LY2_LX0p1_BCbend_euler_Swaveguide_SYNone_SXTNone_SXBNone_Scoupler_CNone_WSTrue_PESo3_PESo4_PECo2_PECo1_N2_W1_L0_R5_ENone_K_empty
* cell instance $2 m0 *1 77.66,0
X$2 2 1
+ mzi_DL5_LY2_LX0p1_BCbend_euler_Swaveguide_SYNone_SXTNone_SXBNone_Scoupler_CNone_WSTrue_PESo3_PESo4_PECo2_PECo1_N2_W1_L0_R5_ENone_K_empty
.ENDS Unnamed_12

* cell mzi_DL10_LY2_LX0p1_BCbend_euler_Swaveguide_SYNone_SXTNone_SXBNone_Scoupler_CNone_WSTrue_PESo3_PESo4_PECo2_PECo1_N2_W1_L0_R5_ENone_K_empty
* pin 
* pin 
.SUBCKT
+ mzi_DL10_LY2_LX0p1_BCbend_euler_Swaveguide_SYNone_SXTNone_SXBNone_Scoupler_CNone_WSTrue_PESo3_PESo4_PECo2_PECo1_N2_W1_L0_R5_ENone_K_empty
+ 1 2
* cell instance $1 r0 *1 0,0
X$1 1 2 coupler_G0p2_L10_D5_D5_W1_L0_ENone
* cell instance $2 r0 *1 15,2.5
X$2 1 bend_euler_W1_R5_L0_ENone_T90_R150
* cell instance $3 r90 *1 24.35,11.85
X$3 1 waveguide_W1000_L2000_L0_ENone
* cell instance $4 r180 *1 33.7,23.2
X$4 1 bend_euler_W1_R5_L0_ENone_T90_R150
* cell instance $5 r0 *1 33.7,23.2
X$5 1 waveguide_W1000_L100_L0_ENone
* cell instance $6 m90 *1 33.96,23.2
X$6 1 waveguide_W1000_L160_L0_ENone
* cell instance $7 r0 *1 57.66,0
X$7 1 2 coupler_G0p2_L10_D5_D5_W1_L0_ENone
* cell instance $8 r270 *1 43.31,11.85
X$8 1 bend_euler_W1_R5_L0_ENone_T90_R150
* cell instance $9 m135 *1 43.31,13.85
X$9 1 waveguide_W1000_L2000_L0_ENone
* cell instance $10 m0 *1 33.96,23.2
X$10 1 bend_euler_W1_R5_L0_ENone_T90_R150
[... 113 lines cut ...]
.ENDS bend_s_W1_H1p9_L5_L0_NP99_TS0_TS1_E8a9ff7cf

* cell straight_coupler_G0p2_L10_W1_L0_E8a9ff7cf
* pin 
* pin 
.SUBCKT straight_coupler_G0p2_L10_W1_L0_E8a9ff7cf 1 2
* cell instance $1 m0 *1 0,0.6
X$1 1 waveguide_W1000_L10000_L0_E8a9ff7cf
* cell instance $2 r0 *1 0,-0.6
X$2 2 waveguide_W1000_L10000_L0_E8a9ff7cf
.ENDS straight_coupler_G0p2_L10_W1_L0_E8a9ff7cf

* cell waveguide_W1000_L10000_L0_E8a9ff7cf
* pin 
.SUBCKT waveguide_W1000_L10000_L0_E8a9ff7cf 1
.ENDS waveguide_W1000_L10000_L0_E8a9ff7cf
```

To define connection stack, we can use the l2n.connect function. 
```
r1 = l2n.make_layer(0, "layer0")
r2 = l2n.make_layer(2, "layer2")
l2n.connect(r1)  # intra layer
l2n.connect(r1, r2)  # inter layer
```

sebastian-goeldi, 2023-04-03:
> Ah I see, understood. So I think whicheber format or netlist extractor we use should have to take into account the connections (how and what is connected to what), and also the device parameters such as radius of curvature.

Yes, well, that can come second, we will not be able to do that in KLayout, I think device extractors have no knowledge of cells, so that will make it tricky. I am not sure whether klayout netlist extraction is the best thing to do for photonics. You don't have the clear distinction as in electronics with "a resistor is if we have two pins and they are connected with a resistive layer characterized by this doping (or no doping) on silicon and it will result in R=width/height*sheet_resistance" (same with transistors as Source/Drain/Gate).

In photonics we care about wavelength, radius, proximity to other material. So, essentially except for straight waveguides, we would need to have compact models and then drop in replace them with [black box circuits](https://klayout.de/doc-qt4/manual/lvs_tweaks.html#h2-147). This creates significant overhead and we are probably better off not using the LVS of KLayout imo.

sebastian-goeldi, 2023-04-03:
> How do we define the connection stack programmatically?

I am not sure how to produce it programatically, for electronic layout you need logical layers (doping & silicon e.g.). The technology file can store it, but I am not sure how to load the layer stack from there

tvt173, 2023-04-04:
Agreed with @sebastian-goeldi . Having written a few different implementations of photonic LVS in the past, I'm not convinced that using an electrical LVS tool for photonics is the best route. It always feels a bit Frankensteined and hacky. I wouldn't want to re-write an electrical LVS tool if Klayout already has something decent, so maybe use that for electrical. But it may be worth it to do something a bit more custom for photonic LVS... and maybe something like the graph comparison I pitched above.

Of course though, it really matters how you define your schematic! Particularly the interplay and distinction between photonic and electrical components and connections. Maybe that's the right place to start...

das-dias, 2023-04-04:
I have no experience in specifically designing an electronic LVS tool, but
I have implemented some netlist extraction functionalities from layout in
https://github.com/das-dias/reds, using an approach based on networkx
that @Troy Tamas mentioned. Having read the problem described by
@sebastian-goeldi, I agree that a dedicated LVS tool for photonics should
be used. If possible, I'd like to team up with you @Troy Tamas to push
forward on this matter, if possible.

tvt173, 2023-04-04:
nice tool @das-dias ! in the past, I had looked for an open source tool that could do parasitic extraction and I was hard-pressed to find something that suited my needs... would love to take a deeper look at this. also to collaborate on this project with you and anyone else interested

SkandanC, 2023-04-04:
Sounds great, I would be like to also be interested. LVS for photonics is an issue that as of now has no clear answer. Would be nice to work out a flow which can support both maybe?

SkandanC, 2023-04-04:
Also last I heard, IEEE was working on a format to describe both electronics and photonic components in the same sort of "netlist". This would be used for electronic-photonic co-simulation. Perhaps that;s also something to look into. We do ahve a dsfactory issue on that too

das-dias, 2023-04-05:
Can you send the link to the IEEE standard you mentioned? I would love to
check it out!

5/04/2023 à(s) 00:03:

sebastian-goeldi, 2023-04-05:
I think Skandan means the one linked in here by @simbilod https://github.com/gdsfactory/gdsfactory/issues/1136

atorkmabrains, 2023-05-04:
@joamatab We have added the net tracing capability as we have discussed for Generic technology here: https://github.com/gdsfactory/gdsfactory/pull/1611

Please check and let me know.

atorkmabrains, 2023-05-04:
@joamatab we haven't finished the LVS rule deck. Could you reopen this issue or create a new one for creating the LVS rule deck for the Generic technology?

atorkmabrains, 2023-05-07:
@joamatab We have added the basic folder structure for LVS rule deck with heater recognition: #1620

atorkmabrains, 2023-05-09:
We could add it under pdk section in GDS Factory.

joamatab, 2023-05-09:
we need an example doc for this and ideally tests too

atorkmabrains, 2023-05-10:
Net Tracing Documentation:

https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/generic_tech/README.md

atorkmabrains, 2023-05-10:
LVS documentation:

https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/generic_tech/klayout/lvs/README.md

atorkmabrains, 2023-05-10:
LVS testing documentation:

https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/generic_tech/klayout/lvs/testing/testcases/README.md

atorkmabrains, 2023-05-10:
Heater LVS testcase:

https://github.com/gdsfactory/gdsfactory/tree/main/gdsfactory/generic_tech/klayout/lvs/testing/testcases/unit/heater_devices

joamatab, 2023-10-24:
@nikosavola

PR #1611 (2023-05-04): Adding net tracing capability in klayout for Generic PDK
Fixes: #934 
Adding net tracing capability in klayout for Generic PDK. Net tracing help debug layouts easily.