gdsfactory/gdsfactory #943: gf tool install issues

joamatab, 2022-12-01:
Hello @joamatab ,

I installed the latest version and did a fresh install of klayout, and tried to use the cli tool to install the salt package. I ran into a couple of issues:

1. The cli tool is broken and does not take in any commands and errors out. I had to manually fix it by opening the gf file and changin `gf` to `cli` everywhere.

[image]

2. After getting this working, I ran `gf tool install` and it was installed in the salt directory. Opening klayout did not show the menu item for gdsfactory:

[image]

Moving the gdsfactory folder to `macros` rather than `salt` brings the menu back:

[image]

Is there a reason we have to keep this under salt? Does it change it's functionality in any way by moving it to macros? Atleast keeping it under macros brings it in toolbar. Having it under salt makes it inaccessible.

Perhaps @thomasdorch knows?

_Originally posted by @SkandanC in https://github.com/gdsfactory/gdsfactory/issues/937#issuecomment-1332682864_

sebastian-goeldi, 2022-12-01:
If you want it installable by the package manager it needs to go to the salt folder, that's where all packages managed by the saltmine live. I believe though you have an error in your setup. Python macros need to go thw `pymacros` folder. `macros` is for ruby. That might be the problem. I think the standard packages might detect this and can run it successfully anyway, but not for packages.

joamatab, 2022-12-04:
Skandan,

is it working now?

sebastian-goeldi, 2022-12-04:
I believe this will not work. Klayout has two kinds of python files. `.py` and `.lym` They get loaded differently. `.py` are the "traditional" python files and go into `/python`. `.lym` are the xml wrapped python files that also hold klayout information. These are the ones called by the macro class (forgot the exact name) and can be set to autoexecute etc. They need tp be in `/pymacros`. I believe if they are in `/python`, they won't be properly executed on startup anymore.

SkandanC, 2022-12-05:
@joamatab yes it works now.

closed 2022-12-05 by joamatab