gdsfactory/gdsfactory #950: store reference gds designs

joamatab, 2022-12-01:
For the difftest what is the best option to store reference files [here](https://github.com/gdsfactory/gdslib/tree/data/gds/gds_ref) ?

Some ideas:

- moving them to GitHub binary artifacts
- database (see https://github.com/gdsfactory/gdsfactory/tree/database/gdsfactory/database)

tvt173, 2022-12-06:
I've played around with a few options in the past, but ultimately I've found the simplest solution has been the best and easiest to maintain: just keep them in the same repo as the code. Usually the component GDS files are not that big and git actually does a pretty decent job of compressing over the history of the repo, even for the binary files, at least on the server side.

Another option that integrates well with git could be [DVC ](https://dvc.org/), but it would take a bit more setup/explanation on the developer's part and you would need to find a place to host it.

flaport, 2022-12-06:
I've been doing this with [dvc](https://dvc.org), but it definitely is a hurdle for new users to get used to. I don't mind checking in GDS files into git into internal repositories, and when there is only a limited number of gds files to check in. For GDSFactory, I expect the number of GDS files to grow quite large over time, so I'm not sure if that's the best idea.

Alternatively, you can also just check them into a public s3 bucket / gcs storage and just keep a datafile to their urls. This might be a little bit lower effort than setting up DVC.

Or just track them with [git-lfs](https://docs.github.com/en/repositories/working-with-files/managing-large-files/configuring-git-large-file-storage)

tvt173, 2022-12-06:
git-lfs is definitely another option to consider... it will solve the issue of the repo, on the _user's_ side, from bloating, which is maybe the main issue here. and it is still a tad more complicated to use than plain git, but not too much

the one thing I noticed when doing it though is that, counterintuitively, the repo on the _server_ side becomes much larger than with plain git, because git compresses over history and git-lfs does not. so if we need to battle storage limits with service providers, plain git is actually the most compact option (but maybe this is not an issue for a public github repo?)

tvt173, 2022-12-06:
looks like storage limits are 1GB? which is not that much... unless you pay for extra storage
https://docs.github.com/en/repositories/working-with-files/managing-large-files/about-storage-and-bandwidth-usage

a separate, dedicated repo, like you have now might be a better solution. and you can flush old history at some point if it becomes problematic

sebastian-goeldi, 2022-12-07:
I see two additional options: 1)  use gzip on the gds (I am not sure what you use for the difftest, but klayout for example natively supports opening `.gds.gz` 2) use oasis which does a better job for compression

closed 2023-11-02 by github-actions[bot]