gdsfactory/gdsfactory #976: Failed installation on Mac with M1 Pro chip

Djordje117, 2022-12-08:
**Describe the bug**
After running `make install`, the installation failed. The summary of the error:
`Collecting tables
  Using cached tables-3.7.0.tar.gz (8.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      /var/folders/8d/48791dp11n91x75y22bzz8_c0000gn/T/H5closeatm6dxyu.c:2:5: error: implicit declaration of function 'H5close' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
          H5close();
          ^
      1 error generated.
      cpuinfo failed, assuming no CPU features: No module named 'cpuinfo'
      * Using Python 3.10.8 | packaged by conda-forge | (main, Nov 22 2022, 08:25:13) [Clang 14.0.6 ]
      * Found cython 0.29.32
      * USE_PKGCONFIG: False
      * Found conda env: ``/Users/dordegluhovic/mambaforge``
      .. ERROR:: Could not find a local HDF5 installation.
         You may need to explicitly state where your local HDF5 headers and
         library can be found by setting the ``HDF5_DIR`` environment
         variable or by using the ``--hdf5`` command-line option.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.`

**To Reproduce**
run `make install` after cloning gdsfactory

joamatab, 2022-12-09:
The issue is coming from https://github.com/PyTables/PyTables/issues/976 not having wheels for macos-arm

it's only one of the simulation plugins optional dependencies

joamatab, 2022-12-12:
You can `pip install -e .`  for the gdsfactory packages

for the pytables dependency, you can try
`pip install git+https://github.com/PyTables/PyTables.git@v3.7.0`

let me know if that works