gdsfactory/gdsfactory #992: database improvements

joamatab, 2022-12-13:
Improve database docs

TODO:

- [ ] add examples on how to add and query measurements and simulations
- [ ] add examples on how to bring up and connect your own database

Maybe:

- [ ] Include GDS serialization (polygons and references)
- [ ] material database

_Originally posted by @joamatab in https://github.com/gdsfactory/gdsfactory/issues/990#issuecomment-1349592302_

Jan-David-Black, 2023-01-20:
How would we deal with development compiles? In a previous work environment of mine, we had a similar system. But we had the problem, that repeated compiles (e.g. for testing) lead to enormous amounts of components in the database, that are completely irrelevant but hard to purge from the DB without affecting interesting data. 
We adopted a `--production` flag. Such a flag could then be used with CI to write to the DB for a taged commit (the tapeout).

joamatab, 2023-01-20:
I think for development one could use a local sqlite version of the database

proppy, 2023-01-26:
Thinks worth exploring:
- SQLite replication to GCS using https://litestream.io/
- JSON streaming to BigQuery https://cloud.google.com/bigquery/docs/loading-data#streaming

joamatab, 2023-02-27:
Niko,

the database notebook example is not running and I had to deactivate its execution

https://gdsfactory.github.io/gdsfactory/notebooks/12_database.html

@nikosavola

proppy, 2023-07-04:
Also saw https://blog.turso.tech/sqlite-based-databases-on-the-postgres-protocol-yes-we-can-358e61171d65 recently.

joamatab, 2023-10-24:
moved to gplugins

closed 2023-10-24 by joamatab