gdsfactory/gdsfactory #1006: material database

joamatab, 2022-12-15:
- How can we leverage refractive index databases available online?
- How can we add your own material refractive index info for a particular PDK?
- how can we fail if we try to add a component material that is not in the database?

- [ ] Document how to add materials for a particular PDK process
- [ ] Decide where to pull the materials from

@simbilod 
@HelgeGehring 
@momchil-flex 
@tylerflex
@tvt173
@thomasdorch 
@SkandanC 
@heitzmann

tvt173, 2022-12-15:
@flaport

flaport, 2022-12-15:
You can probably scrape refractiveindex.info which offers refractive index data in the public domain. I think it's fine to just store these in a public bucket or so.

momchil-flex, 2022-12-15:
In Tidy3D, we need to fit the material data to a model that can be implemented in FDTD. In the fitter that we provide, there's a class method that can initialize it from url linking to a csv file that contains the data. We assume the same file format as refractiveindex.info. https://docs.flexcompute.com/projects/tidy3d/en/latest/_autosummary/tidy3d.plugins.DispersionFitter.html?highlight=dispersion%20fitter

But yeah, scraping the data sounds good to me.

simbilod, 2022-12-15:
The refractiveindex.info website is actually hosted on Github and like Floris said is in the public domain, so we don't even need to scrape it: https://github.com/polyanskiy/refractiveindex.info-database. All index data is tabulated in csv's.

We could add an optional dependency that would pip install the repo, and add a parent OpticalMaterial class under e.g. gdsfactory/database/optical_material.py that has methods to load the files, interpolate index, or provide other transformations like the model fitting from tidy3d (could import from there, but hide it in the class).

Then all solvers can just call gdsfactory.database.materials.optical.MaterialName where MaterialName is a child class of OpticalMaterial. Each solver then uses the same material object in the way that makes sense for it.

If you want to make your own material, you can easily do that by instanciating an OpticalMaterial object with some sampled wavelength, index data.

This would extend nicely to other material data like SemiconductorMaterial, ThermalMaterial, etc. pulled from other sources.

simbilod, 2022-12-15:
Alternatively just use this https://github.com/jtambasco/opticalmaterialspy

lucas-flexcompute, 2022-12-15:
I have a similar script to opticalmaterials.py that might be interesting for you as well:
https://gist.github.com/lucas-flexcompute/5186955dc04dc2d1918cd7c6b5d59d5e

simbilod, 2022-12-16:
I have a little something on a branch https://github.com/simbilod/gdsfactory/tree/material_database

@joamatab any thoughts? It's a one liner to get fitted index data, and even units are handled seamlessly

joamatab, 2022-12-17:
looks good Simon,

i tried to solve both issues in this [PR](https://github.com/gdsfactory/gdsfactory/pull/1007)

closed 2023-01-01 by joamatab