gdsfactory/gdsfactory #1026: tidy3d improvements

joamatab, 2022-12-20:
- [ ] create a shared queue for simulations running on the server
- [ ] make sure grating coupler simulations match meep and lumerical
- [ ] add mesh-refinement option

Maybe:

- [ ] add generators from tidy3d docs for different materials (Si, SiN), wavelengths (1550,1310) and thickness

@momchil-flex 
@tylerflex
@lucas-flexcompute

tylerflex, 2022-12-20:
I like the `Queue` idea. maybe we can include this in the webapi refactor. Is this a new feature in python 3.11?

With the adjoint plugin (testing phase now) we have set things up a little differently right now. We have a few jax-compatible subclasses of regular tidy3d components, which we set up within our `function_to_minimize()`, and then we can use regular `jax.grad` etc. to take gradients of functions that involve a tidy3d simulation run. We can also feed this gradient to a spicy optimizer or other generic optimizer. Basically, I'm not sure if we even need `td.Optimize` so have left it out for now.

Here is an unreleased [example](https://github.com/flexcompute-readthedocs/tidy3d-docs/blob/feature/adjoint/docs/source/notebooks/AdjointPlugin_3_InverseDesign.ipynb) of an inverse design using the plugin. There are also some tutorials in progress [here](https://github.com/flexcompute-readthedocs/tidy3d-docs/blob/feature/adjoint/docs/source/notebooks/AdjointPlugin_1_Intro.ipynb) and [here](https://github.com/flexcompute-readthedocs/tidy3d-docs/blob/feature/adjoint/docs/source/notebooks/AdjointPlugin_2_GradientChecking.ipynb) if you want to learn more about how it is likely to work at this first iteration.

joamatab, 2022-12-20:
examples look great

the queue has been around for a while, i haven't used it yet
https://docs.python.org/2.7/library/queue.html

closed 2023-10-24 by joamatab