gdsfactory/gdsfactory #1049: Mirroring cross-sections

simbilod, 2022-12-28:
In #1048 I added an argument to flip the offsets of all sections in the pn cross-section

What would be the best place to add such a feature to generic cross-sections? Say, an argument in CrossSection that, if True, flips the signs of offset, all section offsets, all cladding_offsets, and all bbox_offsets?

CrossSection does not have a constructor where we could do this. If path.extrude is always called, maybe we could put it here. Does that cover all cases?

joamatab, 2022-12-28:
yes, that would be a great addition

This is on fixme/
```
"""FIXME.

- bends in routes flip the polarity in some cases.

"""

import gdsfactory as gf

if __name__ == "__main__":
    c = gf.Component()

    length = 50
    s1 = c << gf.components.straight_pin(length=length)
    s2 = c << gf.components.straight_pin(length=length)

    s2.move((100, 50))

    route = gf.routing.get_route(
        s1.ports["o2"], s2.ports["o1"], cross_section=gf.cross_section.pin
    )
    c.add(route.references)

    c.show(show_ports=True)
```

[image]

simbilod, 2022-12-28:
#1050 extends the mirroring to all cross-sections

To fix the routing polarity issues: I think we can add a new argument in component.mirror and componentReference.mirror that, if True, makes it so the returned Component or Reference is the base component but with all its ComponentSpec arguments having mirror=True.

Then, when appropriate we can use this flag, for instance in the manhattan function if x_reflection is true.

closed 2023-11-02 by github-actions[bot]