gdsfactory/gdsfactory #1067: SAX interface simulations

simbilod, 2023-01-03:
See [#13](https://github.com/flaport/sax/issues/13) in SAX

How easy would it be for sax.circuit to also account for interfaces between components?

I am thinking a flag that, if True,
(1) requires the user to provide models for component pairs that show up connected in the graph
(2) adds a component evaluating these models at every "connection" of the netlist

The obvious use case is to model reflections in direct transitions between waveguides with different widths or bend radii. Currently gdsfactory+SAX throws a critical warning for the former, and evaluates the second without mode mismatch reflections. 

Maybe this should be added as an option in gdsfactory.get_netlist()?

Example behaviour for bend --> coupler --> straight :

Input:
circuit, _ = sax.circuit(c.get_netlist(), models=models, interfaces=True)

Output:

    "Given Models": [],
    "Required Models": ["bend_circular", "coupler_full", "straight"],
    "Required Interface Models": [("bend_circular", "coupler_full"), ("bend_circular", "straight")],
[If there were open-source EMEs around](https://github.com/flaport/meow) the required reflection matrices would be very easy to calculate :)

tvt173, 2023-02-20:
@simbilod, I would argue that transitions between different cross-sections, like the example you give, should be captured and modeled as their own components. this would get rid of the issue you raise. and would also make it easier to validate that all connections in your circuit are proper. what do you think?

simbilod, 2023-02-20:
> @simbilod, I would argue that transitions between different cross-sections, like the example you give, should be captured and modeled as their own components. this would get rid of the issue you [...]

Yes I think that is probably the best thing to do at the gdsfactory level. There was interest in implementing something like this as an option in SAX directly (see linked issue), but I haven`t had time to wrap it up yet.

Anyway, I will close this for now in this repo

closed 2023-02-20 by simbilod