gdsfactory/gdsfactory #1078: Suggestion: Infer version instead of bumping with bumpversion

nikosavola, 2023-01-04:
I noticed [peritus/bumpversion](https://github.com/peritus/bumpversion) is used for bumping hard-coded version numbers around the repo in specific bump commits.

I think it would be nicer to move away from a tool like this to always inferring the version from the latest semver tag in the repo. For implementation, the CC0-licensed https://github.com/jbweston/miniver could be employed.

At least some of these version numbers should be easy to avoid, such as in the `.github/workflows/release.yml` file, where the version may be inferred from the pushed tag.

joamatab, 2023-01-04:
Thank you for the suggestion Niko

How would that flow allow to bump derivative files?

```
[bumpversion:file:pyproject.toml]

[bumpversion:file:README.md]

[bumpversion:file:Dockerfile]

[bumpversion:file:docs/conf.py]

[bumpversion:file:docs/contribution.md]

[bumpversion:file:gdsfactory/__init__.py]

[bumpversion:file:gdsfactory/config.py]

[bumpversion:file:gdsfactory/cli.py]

[bumpversion:file:.github/workflows/release.yml]

[bumpversion:file:gdsfactory/klayout/grain.xml]

[bumpversion:file:gdsfactory/klayout/pymacros/klive.lym]

[bumpversion:file:gdsfactory/klayout/pymacros/set_menus.lym]

[bumpversion:file:construct.yaml]

[bumpversion:file:install.sh]

[bumpversion:file:install.bat]
```

nikosavola, 2023-01-04:
At least the [Dockerfile can do it interactively](https://stackoverflow.com/questions/49681984/how-to-get-version-value-of-package-json-inside-of-dockerfile) and `.github/workflows/release.yml` could use [docker/metadata-action](https://github.com/docker/metadata-action)

Some files, such as, the README and `contribution.md` don't really need it. But indeed, not sure about construct.yaml and the grain.xml for the Salt package, maybe they need something like bumpversion. One way to have less bumping would be to do manual bumping only for major and minor versions instead of also patches.

Would make sense to categorise the list for ones where bumpversion could be avoided in this issue, and then decide what to do.

joamatab, 2023-10-24:
We moved to tbump

closed 2023-10-24 by joamatab