gdsfactory/gdsfactory #1081: taper_cross_section_linear does not taper cladding

gdspaul, 2023-01-04:
**Describe the bug**
New taper function doesn't handle cladding regions.  Old `taper2` could do this.

**To Reproduce**
```
xc_sin = partial(
    cross_section,
    width=1.0,
    layer=(0,1),
    cladding_layers=[(0,2), (0,3)],
    cladding_offsets=[5, 10]
)

xc_sin_ec = partial(
    xc_sin,
    width=0.2
)

@gf.cell
def demo():

    taper_length = 10

    in_stub_length = 10
    out_stub_length = 10

    c = gf.Component()

    wg_in = c << gf.components.straight(
        length=in_stub_length,
        cross_section=xc_sin_ec
    )

    taper = c << gf.components.taper_cross_section_linear(
        length=taper_length,
        cross_section1 = xc_sin_ec,
        cross_section2 = xc_sin
    )

    wg_out = c << gf.components.straight(
        length=out_stub_length,
        cross_section=xc_sin
    )

    taper.connect("o1", wg_in.ports["o2"])
    wg_out.connect("o1", taper.ports["o2"])

    c.add_port("o1", port=wg_in.ports["o1"])
    c.add_port("o2", port=wg_out.ports["o2"])

    return c
```
[image]

**Expected behavior**
There should be cladding in the taper region.

joamatab, 2023-01-04:
Thank you Paul for reporting this, 

in principle only named Sections were supposed to be tapered, but I understand the convenience of defining them as you did in your code

[image]

gdspaul, 2023-01-04:
Hi Joaquin, thanks for the quick fix.  (This is Paul Hansen btw, hi!)  Should I do something else to get my tapered cladding?  I'm just learning your cool tool.

joamatab, 2023-01-04:
Hi Paul,

yes, I was imagining it was you :) thank you for your issue

I just released `6.18.1` which should work as you expect
i think it makes sense to be able to easily define cladding_layers and cladding_offsets and automatically taper them to the tapers

@simbilod  was also asking for this as well

closed 2023-01-04 by joamatab

gdspaul, 2023-01-04:
hey Joaquin, that screenshot you sent above doesn't look right.  i get something similar in my code.  why doesn't the taper section's cladding match the input and output waveguides' cladding?
[image]