gdsfactory/gdsfactory #1083: add_port can't infer width from cross_section

gdspaul, 2023-01-04:
**Describe the bug**
I want to do this
```
    c.add_port(
        "o2",
        center=xy_snapped[:, -1],
        orientation=0,
        port_type="optical",
        cross_section=cross_section
    )
```
but it raises this exception: `ValueError: Port needs width parameter (um).`  So, I have to do this
```
    c.add_port(
        "o2",
        center=xy_snapped[:, -1],
        orientation=0,
        port_type="optical",
        cross_section=cross_section,
        width=cross_section.width
    )
```

**Suggested fix**
Should be around line 577 of `component.py`.  at least in version 5.49.0 it was.

joamatab, 2023-01-04:
great catch Paul,

this [PR](https://github.com/gdsfactory/gdsfactory/pull/1085) should fix the issue

once the tests pass  can merge it and release a new patch

joamatab, 2023-01-05:
Hi Paul,

yes, thank you for pointing this

I think i finally fixed it in this PR https://github.com/gdsfactory/gdsfactory/pull/1089

closed 2023-01-10 by joamatab