gdsfactory/gdsfactory #1090: `gf.path.smooth` fails with near-collinear points

95-martin-orion, 2023-01-05:
**Describe the bug**
When `gf.path.smooth` is given a sequence of points with `1e-6 < angle < 360 / npoints`, it fails because the `bend` method returns no points.

**To Reproduce**
```python
import gdsfactory as gf
p=gf.path.smooth(points=[(0,0), (0,1000), (1, 10000)])
# IndexError: index -1 is out of bounds for axis 0 with size 0
```

**Expected behavior**
Angles smaller than the interpolation limit should generate at least one point.

**Suggested fix**
Add an `npoints` argument to `gf.path.smooth` and modify this line:
https://github.com/gdsfactory/gdsfactory/blob/578a0d950b8c6829421f13b5ee5881dc9182733b/gdsfactory/path.py#L1295

so that it passes `npoints=max(npoints, int(360 / dt) + 1)` to `bend`. In some adversarial cases this will generate more than one point, but it should always guarantee at least one point is returned.

joamatab, 2023-01-06:
Thank you Orion for the suggested fix
https://github.com/gdsfactory/gdsfactory/pull/1093

I will release gdsfactory 6.18.4 with the fix

@tvt173 and @flaport  have implemented a non-manhattan router that allows you to route diagonal routes, which may be also useful

95-martin-orion, 2023-01-06:
> @tvt173 and @flaport have implemented a non-manhattan router that allows you to route diagonal routes, which may be also useful

Where can I find this router?

joamatab, 2023-01-06:
I think they still need to send a PR for this

closed 2023-01-06 by joamatab