gdsfactory/gdsfactory #1091: Add generic loss function optimiser with simulator support

nikosavola, 2023-01-06:
**Is your feature request related to a problem? Please describe.**

Currently, gdsfactory appears to support SAX and MEEP optimisers. I suggest adding an example or support for a "generic optimisation loop", such that, a user would have to specify their geometry, search space, provide a loss function $\mathcal{L}$, methods for running the simulations and fetching their data.

In photonics, inverse design for individual elements appears popular. This approach is most likely very inefficient for that purpose. Instead, this framework should show how to combine simulation outputs and constraints from arbitrary sources for optimisation. Additionally, the choice of optimiser should be generic as well.

**Describe the solution you'd like**

I've found Ray Tune great for this purpose in not only hyperparamater selection but as a black-box optimiser for physics problems. Additionally, it already supports many other optimiser libraries, including [Ax](https://ax.dev/), [Optuna](https://optuna.org/), and [scikit-optimize](https://scikit-optimize.github.io/stable/). See [Search algorithms](https://docs.ray.io/en/latest/tune/api_docs/suggestion.html) for details.

I'll try providing a prototype example for this.

joamatab, 2023-01-06:
yes Niko, a section of generic optimization sounds like a great idea

we can then apply it to different simulators and components in the docs

nikosavola, 2023-01-16:
Closed after merging d851c46753b0390db6023269418f70605ece452b from #1137.

PR #1137 (2023-01-12): Generic optimiser example using Ray Tune
This PR adds an example notebook for a generic optimiser implementation with Ray Tune.

Closes #1091
closed 2023-01-16 by nikosavola