gdsfactory/gdsfactory #1092: Run linting for notebooks as well

nikosavola, 2023-01-06:
**Is your feature request related to a problem? Please describe.**
It seems linting is currently run in CI with
```bash
flake8 .
```
The numerous notebooks in the repo could be linted as well, though some rules need relaxing.

**Describe the solution you'd like**
Combine a tool like [nbQA](https://github.com/nbQA-dev/nbQA) for linting the notebooks as well as regular code with essentially the same tool (flake8).

joamatab, 2023-01-06:
Hi Niko,

Awesome, yes i love the idea :)

do you want me to send a PR for this?

nikosavola, 2023-01-06:
> do you want me to send a PR for this?

I can also implement this, but if you are itching to do this you can do it as well. I'm just writing down all the features that come to mind as issues.

joamatab, 2023-01-08:
I added it as a pre-commit hook
https://github.com/gdsfactory/gdsfactory/blob/main/.pre-commit-config.yaml

at the moment is running only black

closed 2023-01-11 by joamatab