gdsfactory/gdsfactory #1101: docs issues

joamatab, 2023-01-06:
How could we fix the docs issues for [femwell](https://gdsfactory.github.io/gdsfactory/notebooks/plugins/fem/01_mode_solving.html) ?

simbilod, 2023-01-07:
Looks like it's the Omegaconf not handling numpy floats issue in the femwell caching

simbilod, 2023-01-07:
I was able to fix the first such issue (see my branch [here](https://github.com/simbilod/gdsfactory/tree/femwell_docs)), but now we have numpy floats in the cross-section parameters causing the same issue

[image]

joamatab, 2023-01-07:
yes, there was an issue with omegaconf

now the issue is with the cache,

```
    filtered_layerstack = LayerStack(
        layers={
            k: get_layer_stack().layers[k]
            for k in (
                "core",
                "clad",
                "slab90",
                "box",
            )
        }
    )

    filtered_layerstack.layers["core"].thickness = 0.2

    resolutions = {
        "core": {"resolution": 0.02, "distance": 2},
        "clad": {"resolution": 0.2, "distance": 1},
        "box": {"resolution": 0.2, "distance": 1},
        "slab90": {"resolution": 0.05, "distance": 1},
    }
    lams, basis, xs = compute_cross_section_modes(
        cross_section="rib",
        layerstack=filtered_layerstack,
        wl=1.55,
        num_modes=4,
        order=1,
        radius=np.inf,
        mesh_filename="mesh.msh",
        resolutions=resolutions,
        overwrite=False,
        with_cache=True,
    )
    mode_solver.plot_mode(
        basis,
        np.real(xs[0]),
        plot_vectors=False,
        colorbar=True,
        title="E",
        direction="y",
    )

```

Returns

```
AttributeError: 'numpy.ndarray' object has no attribute 'split'
```

joamatab, 2023-01-10:
Thank you Simon!

closed 2023-01-10 by joamatab