gdsfactory/gdsfactory #1117: meshing plugin does not work with Shapely 2.0.0

joamatab, 2023-01-09:
When updating to Shapely there is an issue with femwell notebook

https://gdsfactory.github.io/gdsfactory/notebooks/plugins/fem/01_mode_solving.html

@simbilod

simbilod, 2023-01-10:
Some of the excessive warning we are seeing when taking intersection of non-overlapping polygons have been fixed with a PR today in shapely dependency:

https://github.com/libgeos/geos/pull/791

joamatab, 2023-01-10:
awesome,

https://github.com/flexcompute/tidy3d/issues/650
https://github.com/flexcompute/tidy3d/issues/642

simbilod, 2023-01-10:
I pushed a small change to mesh plugin in #1127 F

For me everything the notebook runs fine with shapely 2.0

joamatab, 2023-01-10:
Thank you Simon!

i think the issue was with tidy3d

they will release a new version tomorrow

closed 2023-01-10 by joamatab